google-coral/examples-camera #1: detect.py on coral dev-board with gstreamer camera pipeline segfaults

saket424, 2019-05-04:
I tried running the sample . It runs for a little while and the segfaults. Any ideas why ?

mendel@queen-rabbit:/mnt/objd/examples-camera/gstreamer$ python3 detect.py
Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
Detected Edge TPU dev board.
Gstreamer pipeline:  v4l2src device=/dev/video0 ! video/x-raw,format=YUY2,width=640,height=480,framerate=30/1 ! queue max-size-buffers=1 leaky=downstream  ! tee name=t
            t. ! queue max-size-buffers=1 leaky=downstream ! glupload ! glfilterbin filter=glcolorscale
               ! video/x-raw,format=RGBA,width=320,height=180 ! videoconvert ! video/x-raw,format=RGB,width=320,height=180 ! appsink name=appsink sync=false emit-signals=true max-buffers=1 drop=true
            t. ! queue max-size-buffers=1 leaky=downstream ! glupload ! glfilterbin filter=glcolorscale
               ! rsvgoverlay name=overlay ! waylandsink
        
Inference: 43.57 ms FPS: 1.99 fps
Inference: 21.69 ms FPS: 17.06 fps
Inference: 21.45 ms FPS: 16.25 fps
Inference: 21.69 ms FPS: 15.81 fps
Inference: 21.07 ms FPS: 16.52 fps
Inference: 21.61 ms FPS: 14.82 fps
Inference: 21.48 ms FPS: 12.27 fps
Inference: 21.50 ms FPS: 15.17 fps
...
...
Inference: 21.85 ms FPS: 13.53 fps
Inference: 22.00 ms FPS: 16.54 fps
Inference: 21.68 ms FPS: 16.74 fps
Inference: 16.41 ms FPS: 15.95 fps
Inference: 21.20 ms FPS: 19.54 fps
Segmentation fault
mendel@queen-rabbit:/mnt/objd/examples-camera/gstreamer$

saket424, 2019-05-04:
FWIW, I am running the newest 3.0 chef-13 image on the coral compiled and released in April 2019
mendel@queen-rabbit:/mnt/objd/examples-camera/gstreamer$ uname -a
Linux queen-rabbit 4.9.51-imx #1 SMP PREEMPT Fri Apr 5 00:17:46 UTC 2019 aarch64 GNU/Linux

saket424, 2019-05-04:
mendel@queen-rabbit:/mnt/objd/examples-camera$ v4l2-ctl --list-formats-ext --device /dev/video0
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'YUYV'
	Name        : YUYV 4:2:2
		Size: Discrete 640x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 720x480
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.067s (15.000 fps)
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 2592x1944
			Interval: Discrete 0.067s (15.000 fps)
		Size: Discrete 0x0

edgetpu_detect_server --source /dev/video0:YUY2:640x480:30/1 --model all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite --labels all_models/coco_labels.txt --threshold=0.45

The above example does not segfault whereas detect.py does segfault

Other examples here are helpful also https://coral.withgoogle.com/docs/dev-board/camera

mtyka, 2019-05-06:
Hi, I was able to reproduce this, although only sporadically (maybe 1/10 runs). I've pushed a patch to the repo that appears, as far as I can, tell to eliminate the seg fault. Could you try pulling the latest from this repo and trying it again on your board, since it sounds like you were able to get the fault on each launch ? Tnx!

saket424, 2019-05-06:
@mtyka ,
Yes segfault seems eliminated. I'll run it all night to confirm it is indeed fixed. Thanks!

closed 2019-05-06 by saket424

mtyka, 2019-05-07:
Wonderful - thank you for reporting it ! If it starts acting up again, let me know!

Iorek73, 2019-09-16:
Hello all. I'm having a very similiar issue to the above. 

Displaying the face detection on a USB camera runs 100% fine with the following (the export DISPLAY=:0 is necessary to avoid wayland errors)
```
export DISPLAY=:0 && edgetpu_detect \
--source /dev/video1:YUY2:1280x720:20/1  \
--model ${DEMO_FILES}/mobilenet_ssd_v2_face_quant_postprocess_edgetpu.tflite \
--threshold=0.45
```

However, when I try and run my own tflite model  (tflite_paint.tflite), the computer balks. 
```
export DISPLAY=:0 && edgetpu_detect \
--source /dev/video1:YUY2:1280x720:20/1  \
--model ${DEMO_FILES}/tflite_paint.tflite 
```

The warnings are pretty similiar to running face detection, so I'm not too concerned about that.  The screen flickers black for a fraction of a second, then returns to the desktop. 

```
mendel@indigo-snail:/usr/lib/python3/dist-packages/edgetpu/demo$ export DISPLAY=:0 && edgetpu_detect \
> --source /dev/video1:YUY2:640x480:20/1  \
> --model ${DEMO_FILES}/tflite_paint.tflite 
INFO: Initialized TensorFlow Lite runtime.
glvideomixer background=black name=mixer ! glimagesink qos=False sync=False name=glsink
v4l2src device=/dev/video1 ! video/x-raw,format=YUY2,height=480,width=640,framerate=20/1 ! glupload ! tee name=t
t. ! glupload ! queue ! mixer.
overlaysrc name=overlay ! video/x-raw,format=BGRA,height=480,width=640 ! glupload ! queue max-size-buffers=1 ! mixer.
t. ! queue max-size-buffers=1 leaky=downstream ! glfilterbin filter=glcolorscale ! video/x-raw,format=RGBA,height=225,width=300 ! videoconvert ! video/x-raw,format=RGB,height=225,width=300 ! videobox autocrop=True ! video/x-raw,height=300,width=300 ! appsink max-buffers=1 drop=True emit-signals=True sync=False name=appsink
Segmentation fault
```

Iorek73, 2019-10-25:
I have the exact same issue. It cuts after 1 frame. Is it due to the .tflite file being too large?

https://stackoverflow.com/questions/58561680/reducing-tflite-model-size

```
mendel@indigo-snail:/usr/lib/python3/dist-packages/edgetpu/demo$ export DISPLAY=:0 && edgetpu_detect \
> --source /dev/video1:YUY2:1280x720:20/1  \
> --model ${DEMO_FILES}/detect.tflite 

INFO: Initialized TensorFlow Lite runtime.
glvideomixer name=mixer background=black ! glimagesink sync=False name=glsink qos=False
v4l2src device=/dev/video1 ! video/x-raw,height=720,framerate=20/1,format=YUY2,width=1280 ! glupload ! tee name=t
t. ! glupload ! queue ! mixer.
overlaysrc name=overlay ! video/x-raw,height=720,width=1280,format=BGRA ! glupload ! queue max-size-buffers=1 ! mixer.
t. ! queue max-size-buffers=1 leaky=downstream ! glfilterbin filter=glcolorscale ! video/x-raw,height=168,width=300,format=RGBA ! videoconvert ! video/x-raw,height=168,width=300,format=RGB ! videobox autocrop=True ! video/x-raw,height=300,width=300 ! appsink max-buffers=1 sync=False emit-signals=True drop=True name=appsink
Segmentation fault
```

alexanderfrey, 2020-09-01:
```
python3 /home/mendel/google-coral/examples-camera/gstreamer/classify.py --videosrc /dev/video1
Unable to init server: Could not connect: Connection refused

(classify.py:6429): Gtk-WARNING **: 10:52:41.904: Locale not supported by C library.
	Using the fallback 'C' locale.
Unable to init server: Could not connect: Connection refused
Loading ../all_models/mobilenet_v2_1.0_224_quant_edgetpu.tflite with ../all_models/imagenet_labels.txt labels.
Detected Edge TPU dev board.
Gstreamer pipeline:
 v4l2src device=/dev/video1 ! video/x-raw,width=640,height=480,framerate=30/1 ! decodebin ! glupload ! tee name=t
            t. ! queue ! glfilterbin filter=glbox name=glbox ! video/x-raw,format=RGB,width=224,height=224 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
            t. ! queue ! glsvgoverlaysink name=overlaysink
        

(classify.py:6429): Gtk-CRITICAL **: 10:52:42.636: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(classify.py:6429): Gtk-CRITICAL **: 10:52:42.636: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(classify.py:6429): Gtk-CRITICAL **: 10:52:42.636: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
Segmentation fault

```

MrLiem, 2020-10-04:
> Similar issue is occuring here. Already tried coral camera vs external usb camera....
> 
> ```
> python3 /home/mendel/google-coral/examples-camera/gstreamer/classify.py --videosrc /dev/video1
> Unable to init server: Could not connect: Connection refused
> 
> (classify.py:6429): Gtk-WARNING **: 10:52:41.904: Locale not supported by C library.
> 	Using the fallback 'C' locale.
> Unable to init server: Could not connect: Connection refused
> Loading ../all_models/mobilenet_v2_1.0_224_quant_edgetpu.tflite with ../all_models/imagenet_labels.txt labels.
> Detected Edge TPU dev board.
> Gstreamer pipeline:
>  v4l2src device=/dev/video1 ! video/x-raw,width=640,height=480,framerate=30/1 ! decodebin ! glupload ! tee name=t
>             t. ! queue ! glfilterbin filter=glbox name=glbox ! video/x-raw,format=RGB,width=224,height=224 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
>             t. ! queue ! glsvgoverlaysink name=overlaysink
>         
> 
> (classify.py:6429): Gtk-CRITICAL **: 10:52:42.636: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
> 
> (classify.py:6429): Gtk-CRITICAL **: 10:52:42.636: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
> 
> (classify.py:6429): Gtk-CRITICAL **: 10:52:42.636: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
> Segmentation fault
> ```

@alexanderfrey  have you already had the solution for this problem, please help me!!! Thank you in advance.