google-coral/examples-camera #3: view feed on Coral Dev Board

JVTEAM, 2019-06-05:
Given that the dev board has no visual desktop, how is this stream visible to the user?

PeterMalkin, 2019-06-05:
You have two options:
1) Plug in a monitor through HDMI cable to get a visual desktop environment
`edgetpu_demo --device`

2) Use streaming mode to run the demo
`edgetpu_demo --stream`

More info: https://coral.withgoogle.com/docs/dev-board/get-started/#run-a-model-on-the-edge-tpu

closed 2019-06-05 by PeterMalkin

JVTEAM, 2019-06-06:
That runs the car demo.  It doesn't stream this particular script....

PeterMalkin, 2019-06-06:
I am not sure what you are asking. But the code is open source. Feel free to modify it to make it do what your use case requires.

JVTEAM, 2019-06-06:
That code doesn't play anything but the edgetpu demo.

JVTEAM, 2019-06-07:
I'm trying to stream the classify.py etc.  The demo stream doesn't do that. Just the demo "car stream"