google-coral/examples-camera #4: How can I see the result when running detect.py

pharrellyhy, 2019-06-13:
Hi,

I'd like to use the coral camera to do realtime detection and see the result on the screen at the same time. But after running detect.py, nothing showed on the screen. Any help would be appreciated. Thanks!

mtyka, 2019-07-08:
Can you provide more details ? Which device are you running this on ?
What does the console show (where you executed the example) ?

ColinWang1203, 2019-07-15:
Running a  coral dev board here and went through the camera setup guide.
I can see my USB camera video on the streaming web on http://192.168.100.2:4664/.

But after trying this(run the detect.py) there is no output on the terminal and no file is produced?

mtyka, 2019-07-19:
If you're running the gstreamer/detect.py in this repo and you're using a USB camera on the dev board instead of the Coral camera, you need to point gstreamer at /dev/video1 instead of /dev/video0 (whether or not the coral camera is attached). Edit this line:
https://github.com/google-coral/examples-camera/blob/89a944ac4f5a8025b735babd929f957e9e3914aa/gstreamer/gstreamer.py#L64
and set it to /dev/video1

If that doesn't resolve the issue, could please post your full command line and console output here so I can debug this ? Also what does:
v4l2-ctl --device=/dev/video0
and
v4l2-ctl --device=/dev/video1

output ?

closed 2019-07-30 by mtyka