google-coral/examples-camera #9: Error in gstreamer based sample code to classify images

Py-JaC, 2019-07-29:
Hi All
I want to run gstreamer pipeline to get images and perform custom actions on them. I saw the provided sample code in gstreamer to classify, detect objects and I tried to run that on my google coral dev board but I am getting errors. 
I am running it on coral camera.
Following is the terminal output I am getting :
 

> In Line 94 of gstreamer.py in run_pipeline

> pipeline = Gst.parse_launch(pipeline)

> GLib.Error : gst_parse_error : no element queue (1)

Please note : Before trying the example script I installed gstreamer-1.16 from the source. It was a mistake. But then I unistalled it and tried as mentioned in the example scripts on this repo.

 I suspect that my installing gstreamer on my own broke the existing installation but I am not able to figure out how to correct it now. I tried unistalling all the related packages and installing it again but I am getting same error as mentioned above. I don't want to flash the os again. 

Any help will be appreciated. I am stuck on this
Thanks

mtyka, 2019-07-30:
You may have left gstreamer in a strange half installed situation. I would recommend backing up your home directory to another machine and reflashing your board using the instructions here: https://coral.withgoogle.com/docs/dev-board/get-started/#flash-the-board
It's pretty easy to do. After that the example should work (after running install_dependencies.sh). Gstreamer is already installed, the install script will just add svgwrite.

closed 2019-07-30 by mtyka

Py-JaC, 2019-08-04:
Thanks. That worked