google-coral/examples-camera #11: Not able to get 30 fps in 1920x1080 resolution using gstreamer

Py-JaC, 2019-08-19:
Hi
I wanted to read frames at 30 fps at 1920x1080 resolution. I am currently using google coral camera which supports this resolution at 30fps.
In the gstreamer code provided in the repo I removed all frame processing to just receive the frames and see the speed but I am getting only 10 fps. 30 fps is really critical for my application so please suggest why is this so. I tried opencv VideoCapture also but that is giving errors like Resource not available. 
I would prefer to use gstreamer and ideally it should support. Why so many frames are getting dropped in 1920x1080 ?
It works well at 640x480.

mtyka, 2019-08-22:
Hi, there are several issues at hand here, which i'll try and address. The TL;DR is: 
1) The camera yields YUY2, not RGB which likely is what you want for your neural net. Conversion can be slow.
2) To get maximum speed don't use Python, use C(++).

Details:
Whether end-to-end 1080p@30 is supported or not really depends on what you want to do with the frames, what color format you can work with and your run time environment requirements.

The camera, like most cameras, produce a planar color format. In this case YUY2 (4:2:2 chroma subsampling). The camera puts the pixel data in physically contiguous buffers allocated by the ION memory allocator. Getting CPU access (mapping) to the pixels is very fast in native code and slightly slower in Python since the latter needs to do extra steps to expose the underlying native memory to interpreted code. A C/C++ app will thus be faster here. Once the pixel data is mapped the bottleneck becomes what you actually do it. If you need to access every pixel, e.g. copy the frame to a different buffer, it won't do 1080p@30 since CPU processing is slow and frame size is big. 

Working with planar formats isn't always the best option so typically one wants to convert to a RGB format. Doing this conversion on the CPU is slow so 1080p@30 isn't possible without leveraging the GPU. In that case one needs to upload the frame as a GPU texture and do the conversion there. If CPU access to pixel data is then desired the pixels must be downloaded from the GPU again. This is generally also slow but on the dev board there's a GPU specific fast path if the download format is RGBA (not RGB, i.e. with alpha channel). If the goal is to get RGBA data from the camera into a Python application 1080p@30 is possible (depending on the processing actually done with the data in the application).

The examples-camera/gstreamer example code does a couple of things that may not be needed depending on use case.
The camera frames are tee'd to two branches, inference and rendering.
For inference the the camera frame is first downscaled to from 640x480 (YUY2) to 320x180 (RGBA) so that the fast path download is used. Then on the CPU it's converted to 320x180 (RGB) using videoconvert, since the model used doesn't expect an alpha channel. This tee branch could have started with 1080p frames and still hit 30 fps since all CPU access is done on very small frames.
For rendering it draws an SVG overlay on top of the camera frame. This is done on the CPU ("software rendering") and is therefore slow for higher resolutions. Drawing with rsvgoverlay also requires BGRA, so the above download fast path isn't utilized after converting from YUY2 on the GPU.

The purpose of examples_camera is to keep example code short and simple while still providing visual feedback and cross-device compatibility - it's not optimized for speed. If you want to play with the above insights, try the code below. You can see as soon as you un-comment the numpy buffer line, your FPS drops from 30 to 10FPS. 
Switching to C++ would make this somewhat faster, but ultimately, touching/moving/copying every pixel of a 1080p image on CPU will fundamentally not result in a high speed pipeline.

```
import time
import gi
import numpy as np 

gi.require_version('Gst', '1.0')
gi.require_version('GstBase', '1.0')
from gi.repository import GLib, GObject, Gst, GstBase
GObject.threads_init()
Gst.init(None)

def on_new_sample(sink):
    sample = sink.emit('pull-sample')
    buf = sample.get_buffer()
    result, mapinfo = buf.map(Gst.MapFlags.READ)
    if result:
        # Pixel data now available for reading in mapinfo.data
        # What's done with the data here will affect fps.
       
        # Uncomment to get RGB buffer into numpy (will reduce FPS!)
        #rgba_img = np.frombuffer(mapinfo.data, np.uint8)
        #rgba_img = np.reshape(rgba_img, [1080, 1920,  -1])
        #rgb_img = rgba_img[:,:,0:3]

        buf.unmap(mapinfo)
    return Gst.FlowReturn.OK

def on_fps_measurements(element, fps, droprate, avg, appsink):
    caps = appsink.get_static_pad('sink').get_current_caps()
    print('fps: {:.2f} avg: {:.2f} caps: {}'.format(fps, avg, caps))

def main():
    PIPELINE = """v4l2src ! video/x-raw,format=YUY2,width=1920,height=1080 ! 
        {leaky_q} ! {convert} ! {sink}
    """
    FPSSINK = 'fpsdisplaysink name=fps video-sink="appsink name=appsink"'
    LEAKY_Q = 'queue max-size-buffers=1 leaky=downstream'
    
    # To explore differetn conversions, uncomment *only* one of these lines at the time.
    # 1. YUY2 bufers into appsink
    #CONVERT = 'identity'
    # 2. RGBA buffers into appsink
    CONVERT = 'glupload ! glcolorconvert disable-passthrough=true ! gldownload ! video/x-raw,format=RGBA'
    # 3. RGB buffers into appsink
    #CONVERT = 'glupload ! glcolorconvert disable-passthrough=true ! gldownload ! video/x-raw,format=RGB'
    pipeline = PIPELINE.format(leaky_q=LEAKY_Q, convert=CONVERT, sink=FPSSINK)
    print(pipeline)
    pipeline = Gst.parse_launch(pipeline)
    appsink = pipeline.get_by_name('appsink')
    appsink.set_property('emit-signals', True)
    appsink.connect('new-sample', on_new_sample)
    fps = pipeline.get_by_name('fps')
    fps.set_property('text-overlay', False)
    fps.set_property('signal-fps-measurements', True)
    fps.connect('fps-measurements', on_fps_measurements, appsink)
    # Run pipeline.
    pipeline.set_state(Gst.State.PLAYING)
    try:
        GObject.MainLoop().run()
    except:
        pass
    # Clean up.
    pipeline.set_state(Gst.State.NULL)
    while GLib.MainContext.default().iteration(False):
        pass

if __name__ == '__main__':
    main()
```
[Huge thank you to Jonas Larsson for this detailed answer]

closed 2019-08-22 by mtyka

Py-JaC, 2019-08-23:
I really appreciate your detailed reply. Thanks and I will try the code given by you.

jonasl, 2019-11-27:
FYI this has been significantly optimized in the latest system image for the dev board, and the sample code updated to reflect. When you run the latest version of both, classify and detect will run at 1080p on Coral Dev board at 30 fps.

Py-JaC, 2019-11-27:
Hey Jonas
Thanks for the update. Really helpful. I will update it and try.

joshfox10, 2022-02-08:
It appears the image format is a major drag on the system with RGB being the preferred format.  I pulled up the datasheet for OV5645 which is the sensor specified in the Google Coral camera datasheet.  The OV5645 datasheet specifies support for output formats: RAW RGB,RGB565/555/444, YUV422/420, YCbCr422.  Is there somewhere within GStreamer where the YUY is defined as opposed to RGB?  Thank you.