google-coral/examples-camera #12: examples-camera

kflmiami420, 2019-08-19:
Is there a section for the Coral-enviromental-sensor-board

mtyka, 2019-08-21:
Information on the environmental sensor board is here: https://coral.withgoogle.com/products/environmental

closed 2019-08-21 by mtyka

kflmiami420, 2019-08-21:
thank you  .    Any chance google can create a github repo for the
enviromental sensor board ?

mtyka, 2019-08-23:
We don't yet have an supplemental code/examples. Are you looking for something specific ?

kflmiami420, 2019-08-26:
Hello

Thank you for replying  , I have started a post on my favorite forum since
I am using the Raspberry Pi plattform
https://www.raspberrypi.org/forums/viewtopic.php?f=45&t=249302    I would
like a script to make enviro_demo.py run auto on boot up  when user runs an
autoboot.sh install bash.

I am looking to connect one or two more sensors via I2C  one VEML 6075 and
one BMP388   one to get UV index one to get altitude
I am looking to connect a pms5003 particle sensor via the UART plug  and
the positive form the analog plug 5v to drive it
and finally to connect an analog device to the analog port using the signal
wire and negative ( a Wind anemometer powered by a 16volt external PS.)

I have the code in python already for the VEML6075 and BMP388 the code for
the pms5003 already exists online and I also have the python code for a
mcp3008 ADC to the windanemometer which will need to be ported to the
Coral's board onboard ADC channel.

Maybe the code could be made  to read from all these other external sensord
and displayed in the onboard SSD1306 128x32 OLED display and automatically
scroll from sensor to sensor .

I dont know how to use the PWM port or even what device would use it ?

maybe incorporate the led when reading the temp sensor

or using the button to shut down the pi safely

thank you for your time   hope the link and samples are usefull for
creating libraries for the board .

I guess when I get all the sensors working I will focus on IOT Cloud
gathering and sharing and collecting side of the project