google-coral/examples-camera #14: Inference on pictures instead of videostream with Coral Camera on EdgeTPU dev board

EspenTeigen, 2019-09-09:
Hi.

I want to run classification inference on pictures triggered from a digital pin on the dev board. But i cant figure out how i can take one picture and send it to the tf-model with the gstreamer library.

mtyka, 2019-09-11:
Take a look at this example project: https://github.com/google-coral/project-teachable
which does something almost exactly as you describe. 
Basically in the callback triggered by a particular frame from gstreamer you can check the GPIO status of your desired pin and then take an action depending that state - e.g. save the image or do whatever. If you want to avoid triggering multiple consecutive frames (say it's a button you want to press to take a picture) you may have to debounce the button state and detect the up or down edge by saving the previous state somewhere.

closed 2019-09-11 by mtyka

EspenTeigen, 2019-09-11:
Thank you so much.

Mvh
Espen

Closed #14<https://github.com/google-coral/examples-camera/issues/14>.