google-coral/examples-camera #19: Error in detect.py

PjotrG, 2019-11-24:
get by running detect.py:
pi@raspberrypi:~/Downloads/examples-camera-master/gstreamer $ python3 detect.pyTraceback (most recent call last):
  File "detect.py", line 35, in <module>
    import gstreamer
  File "/home/pi/Downloads/examples-camera-master/gstreamer/gstreamer.py", line 20, in <module>
    gi.require_version('Gst', '1.0')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 118, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace Gst not available
no Gst.py in /usr/lib/python3/dist.packages/gi/overrides (though GLib.py,Gtk.py ...present)
raspistill -t 0 no display, though produces correct .jpg file (camera working)
how come? how can I fix it? (got program running on another pi)

jonasl, 2019-11-27:
You're missing the GStreamer python bindings.
sudo apt-get install -y gstreamer1.0-plugins-bad gstreamer1.0-plugins-good python3-gst-1.0 python3-gi

See also install_requirements.sh mentioned in the README

PjotrG, 2019-11-28:
you are right, in detail python-gst-1.0 is the problem (the other 3 are o.k.)

what I tried so far:

- deleted all folders python3-gst-1.0 there are(4), tried again no success

- copied python3-gst-1.0 manually no success

 

still the same error message?!

(solved in the meantime the raspistill -t 0 issue, that's working now)

 
 

Gesendet: Mittwoch, 27. November 2019 um 18:24 Uhr

PjotrG, 2019-11-28:
@ jonasl

update:
I solved this issue by deinstall/ install pip3 ... now install_requirements work:
pi@raspberrypi:~/Downloads/examples-camera-master/gstreamer $ sudo bash install_requirements.sh
Reading package lists... Done
Building dependency tree       
Reading state information... Done
gstreamer1.0-plugins-bad is already the newest version (1.10.4-1).
gstreamer1.0-plugins-good is already the newest version (1.10.4-1).
python3-gi is already the newest version (3.22.0-2).
python3-gst-1.0 is already the newest version (1.10.4-1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: svgwrite in /usr/local/lib/python3.5/dist-packages (1.3.1)
Requirement already satisfied: pyparsing>=2.0.1 in /usr/lib/python3/dist-packages (from svgwrite) (2.1.10)
Installing Raspberry Pi specific dependencies
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-rpi.gpio is already the newest version (0.6.5~stretch-1).
0 upgraded, 0 newly installed, 0 to remove and 2 not upgraded.

but by running detect.py I get:
pi@raspberrypi:~/Downloads/examples-camera-master/gstreamer $ sudo python3 detect.py
Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
Gstreamer pipeline:  v4l2src device=/dev/video0 ! video/x-raw,width=640,height=480,framerate=30/1 ! queue max-size-buffers=1 leaky=downstream  ! tee name=t
            t. ! queue max-size-buffers=1 leaky=downstream ! videoconvert ! videoscale ! video/x-raw,format=RGB,width=320,height=180 ! appsink name=appsink sync=false emit-signals=true max-buffers=1 drop=true
            t. ! queue max-size-buffers=1 leaky=downstream ! videoconvert
               ! rsvgoverlay name=overlay ! videoconvert ! ximagesink
            
Traceback (most recent call last):
  File "detect.py", line 99, in <module>
    main()
  File "detect.py", line 96, in main
    result = gstreamer.run_pipeline(user_callback)
  File "/home/pi/Downloads/examples-camera-master/gstreamer/gstreamer.py", line 94, in run_pipeline
    pipeline = Gst.parse_launch(pipeline)
**GLib.Error: gst_parse_error: no element "v4l2src" (1)**

PjotrG, 2019-11-28:
sorry I messed it all up, hope you can still read it

PjotrG, 2019-11-29:
add:
on pi@raspberrypi:~ $ sudo apt-get install gstreamer1.0
I get:
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'gstreamer1.0-videosink' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-alsa' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-ugly-dbg' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-ugly-doc' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-base-apps' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-rtsp-dbg' for regex 'gstreamer1.0'
Note, selecting 'libgstreamer1.0-dev' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-tools' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-good' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-qapt' for regex 'gstreamer1.0'
Note, selecting 'libgstreamer1.0-0-dbg' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-espeak' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-omx' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-nice' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-omx-rpi-dbgsym' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-omx-rpi' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-omx-rpi-config' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-videosource' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-crystalhd' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-fluendo-mp3' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-rtsp' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-bad' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-base' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-audiosource' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-gnonlin-dbg' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-gnonlin-doc' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-really-bad' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-clutter' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-pulseaudio' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-pocketsphinx' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-bad-dbg' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-bad-doc' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-visualization' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-omx-generic' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-lame' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-x' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-omx-generic-dbgsym' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-gnonlin' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-vaapi-doc' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-omx-generic-config' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-libav' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-base-dbg' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-good-dbg' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-base-doc' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-good-doc' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-dvswitch' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-clutter-3.0' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-vaapi' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-libav-dbg' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-packagekit' for regex 'gstreamer1.0'
Note, selecting 'libgstreamer1.0-0' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-omx-dbg' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-omx-bellagio-config' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-ffmpeg' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-plugins-ugly' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-doc' for regex 'gstreamer1.0'
Note, selecting 'gstreamer1.0-audiosink' for regex 'gstreamer1.0'
gstreamer1.0-alsa is already the newest version (1.10.4-1+deb9u1).
gstreamer1.0-libav is already the newest version (1.10.4-1).
gstreamer1.0-plugins-bad is already the newest version (1.10.4-1).
gstreamer1.0-plugins-base is already the newest version (1.10.4-1+deb9u1).
gstreamer1.0-plugins-good is already the newest version (1.10.4-1).
gstreamer1.0-tools is already the newest version (1.10.4-1).
gstreamer1.0-x is already the newest version (1.10.4-1+deb9u1).
libgstreamer1.0-0 is already the newest version (1.10.4-1).
gstreamer1.0-omx is already the newest version (1.10.4-1+rpt3).
gstreamer1.0-omx-bellagio-config is already the newest version (1.10.4-1+rpt3).
gstreamer1.0-omx-generic is already the newest version (1.10.4-1+rpt3).
gstreamer1.0-omx-generic-config is already the newest version (1.10.4-1+rpt3).
gstreamer1.0-omx-generic-dbgsym is already the newest version (1.10.4-1+rpt3).
gstreamer1.0-omx-rpi is already the newest version (1.10.4-1+rpt3).
gstreamer1.0-omx-rpi-config is already the newest version (1.10.4-1+rpt3).
gstreamer1.0-omx-rpi-dbgsym is already the newest version (1.10.4-1+rpt3).
Some packages could not be installed. This may mean that you have
requested an impossible situation or if you are using the unstable
distribution that some required packages have not yet been created
or been moved out of Incoming.
The following information may help to resolve the situation:

**The following packages have unmet dependencies:
 gstreamer1.0-omx-dbg : Depends: gstreamer1.0-omx (= 1.0.0.1-0+rpi12+jessiepmg) but 1.10.4-1+rpt3 is to be installed
E: Unable to correct problems, you have held broken packages.**

How can I fix those?

closed 2019-12-03 by PjotrG