google-coral/examples-camera #22: Can't open camera using webcam logitech C310

alan303138, 2019-11-28:
INFO:OpenGL.acceleratesupport:No OpenGL_accelerate module loaded: No module named 'OpenGL_accelerate'

Can the dev board install OpenGL_accelerate?
THX!

mtyka, 2019-11-29:
Is this on a DevBoard ? If so, note that your USB cam will show up as camera 1 not camera 0. 
Do if you're using gstreamer, set the source to /dev/video1 , if you;re using pygame, change the camera line to ```pygame.camera.Camera(camlist[1], ...```

xav12358, 2019-12-10:
Hello, I get the same trouble on my coral:

`python3 detect.py
Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
Detected Edge TPU dev board.
Gstreamer pipeline:
 v4l2src device=/dev/video1 ! video/x-raw,width=640,height=480,framerate=20/1 ! glupload ! tee name=t
            t. ! queue ! glfilterbin filter=glbox name=glbox ! video/x-raw,format=RGB,width=300,height=300 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
            t. ! queue ! glsvgoverlaysink name=overlaysink
        

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.056: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.060: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.060: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.060: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.060: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.061: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.062: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.062: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.062: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.062: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.062: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.062: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.062: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.063: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.064: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.064: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.064: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.064: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.065: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.065: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed
[... 73 lines cut ...]
(detect.py:3952): Gdk-CRITICAL **: 12:51:45.185: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.185: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.185: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.185: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.186: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.186: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.186: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.186: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.186: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.186: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.186: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:3952): Gdk-CRITICAL **: 12:51:45.186: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed
Segmentation fault
`

mtyka, 2019-12-14:
But it works with a different camera ? 
Check your camera's capabilities using v4l2:
v4l2-ctl --list-formats-ext -d /dev/video1

closed 2020-02-06 by mtyka