google-coral/examples-camera #23: Different results for detect.py with gstreamer, pygame and opencv

PjotrG, 2019-12-09:
@mtyka just for interest: I get very different results for detect.py running the 3 programs
       gstreamer: very stable and consistent
       pygame and opencv: mediocre
I'm using a raspi3 modelB with a picamera, Opencv 3.4.1, Python3(stretch)
and the model mobilenet_ssd_v1_coco_quant_postprocess_edgetpu.tflite  in all cases (v2 doesn't work at all).
Is there a reason for this? Can I improve this (personally I prefer opencv to gstreamer)

PjotrG, 2019-12-11:
I found the problem: the color conversion in open cv from BGR to RGB is missing.
At least when you add cv2_im = cv2.cvtColor(cv2_im, cv2.COLOR_BGR2RGB) it detects a banana to 98% (you then have to convert it back, before imshow to get the right display, there is probably a more elegant way to do this ??)

mtyka, 2019-12-12:
Yikes, thank you for catching that - this burns me every time, I always
forget CV likes BGR default. Wanna submit a pull request ?

-- sent from my phone

mtyka, 2019-12-14:
Fixed in https://github.com/google-coral/examples-camera/commit/16b54cd238bf74ab3d6bcca952b48961bb6285f2.

closed 2019-12-14 by mtyka