google-coral/examples-camera #28: edgetpu_detect_server does not work with other videos

nopphons, 2019-12-18:
I am in the process of setting up my coral. Running edgetpu_demo --stream works fine for me.
There are 2 parts within edgetpu_demo - stream and device. It seems that there are 2 separate videos for testing as well - video_device.mp4 and video_stream.mp4. I looked into both videos and rarely saw any differences (differences I saw include HD vs SD and the dimensions).

Here is the source code for edgetpu_demo - https://coral.googlesource.com/edgetpudemo/+/refs/heads/master/edgetpu_demo for reference

I tried to run edgetpu_detect_server with other videos and it didn't work out as I thought. It gave me a black screen. Does anyone know why this happens? What are the requirements for the videos that I can run with edgetpu_detect_server?

alexanderfrey, 2019-12-19:
Hi, yes there are specific requirements for the videos which are not documented anywhere. I figured that inspecting the videos with ffprobe reveals the settings that are necessary for your own videos. I can not remember those settings from the top of my head but I found them with ffprobe... Good luck !

nopphons, 2019-12-19:
@alexanderfrey Hi Alexander, thanks so much for your help. So you recommend me using ffprobe to check the settings of the provided sampels video that works right? Would it be possible to convert the settings of my videos to satisfy the settings required though? Is there an easy way to do that on the top of your head. Really appreciate your help.

alexanderfrey, 2020-01-02:
Sorry for not answering. Yes, compare the results with ffprobe and convert your videos with those settings with ffmpeg. Easiest way is def. ffprobe and ffmpeg. Good luck !

DannyDeusAI, 2020-01-17:
ffmpeg -i /path/input.mp4 \
	 -an -vf "scale=960:540,format=yuv420p" \
	 -colorspace bt470bg -color_range tv -color_primaries smpte170m \
	 -color_trc bt709 \
	 -vcodec libx264 \
	 -brand mp42 -c:v libx264 -profile:v baseline \
	 /path/output.mp4

worked for me, you may want to adjust to your own need

Namburger, 2020-02-25:
Hi guys, this will also works but a little slower since it involves gstreamer to stream the entire video into the new format (not sure if I'm explaining this correctly):
```
$ gst-launch-1.0 -v filesrc location=/path/to/input.mp4 ! qtdemux ! h264parse ! vpudec ! glfilterbin filter=glcolorscale ! 'video/x-raw,width=960,height=540' ! x264enc key-int-max=30 tune=zerolatency speed-preset=fast threads=4 bitrate=2000 ! video/x-h264,profile=baseline ! mp4mux ! filesink location=/path/to/output.mp4
```

closed 2020-06-12 by Namburger