google-coral/examples-camera #29: Installing OpenCV Error

Wichtel02, 2019-12-22:
Everytime I try to install OpenCV with the given Command pip3 install Pillow opencv-python opencv-contrib-python
I get the Error:

Collecting Pillow
Using cached https://files.pythonhosted.org/packages/5b/bb/cdc8086db1f15d0664dd22a62c69613cdc00f1dd430b5b19df1bea83f2a3/Pillow-6.2.1.tar.gz
Collecting opencv-python
Could not find a version that satisfies the requirement opencv-python (from versions: )
No matching distribution found for opencv-python

I use the Google Coral Dev Board.

Hopefully somebody can tell me how I can fix this problem

bomanimc, 2020-01-07:
I've also been running into this issue.

mtyka, 2020-01-10:
The package you need is ```python3-opencv```

However, the easiest way is just to run this install_script in this repo (under opencv) which should work out of the box on devboard.
https://github.com/google-coral/examples-camera/blob/master/opencv/install_requirements.sh

```bash opencv/install_requirements.sh```

And you should be good to go.

mtyka, 2020-01-10:
Please feel free to reopen this bug if the issue persists.

closed 2020-01-10 by mtyka