google-coral/examples-camera #30: gstreamer examples won't run – tflite_runtime issue

mehmandarov, 2019-12-30:
Hi all,

Here is a description of the issue:
1. Followed installation instructions in readme for gstreamer examples. 
2. Running the code `classify` or `detect` fails with an error message that `tflite_runtime` is missing. 
3. Tried following [instructions](https://www.tensorflow.org/lite/guide/python) for installing `tflite_runtime` for Python 3.5 (that is installed on Coral). The installation went fine, but running `python3 classify.py` fails with a new error:

```
AttributeError: /usr/lib/aarch64-linux-gnu/libedgetpu.so.1: undefined symbol: tflite_plugin_destroy_delegate
```
Thanks!

mtyka, 2020-01-01:
Did you install the edgetpu runtime with 
sudo ```apt-get install libedgetpu1-std``` or ```sudo apt-get install libedgetpu1-max```  ?

In general follow the instructions here: https://coral.ai/docs/accelerator/get-started/

mehmandarov, 2020-01-06:
Yes, the standard package seems to be installed. Btw, I am using the coral dev board.

mtyka, 2020-01-10:
If you're using a dev board you don't need to do  apt-get install libedgetpu1-std.
Reflash your board clean with Mendel *Day* (follow the instructions at https://coral.ai/docs/dev-board/get-started/). Specifically, once flashed you'll need to install the tflite-runtime like so:

```
wget https://dl.google.com/coral/python/tflite_runtime-1.14.0-cp37-cp37m-linux_aarch64.whl
pip3 install tflite_runtime-1.14.0-cp37-cp37m-linux_aarch64.whl
```

mtyka, 2020-01-10:
Please feel free to reopen this bug if the issue persists.

closed 2020-01-10 by mtyka