google-coral/examples-camera #31: memory leak

oliver9523, 2020-01-09:
It looks like there's a memory leak.
I'm using the USB accelerator on a raspberry pi 4, python 3.7.
When I run [detect.py](https://github.com/google-coral/examples-camera/blob/master/opencv/detect.py) the memory usage creeps up.

[image]

- Left side of the plot you can see it climb over time with the example code
- Right side when I comment out the following lines...
_(Note: the drop is a gap in the logs between stopping, commenting out lines starting again)_
```
common.set_input(interpreter, pil_im)
interpreter.invoke()
objs = get_output(interpreter, score_threshold=args.threshold, top_k=args.top_k)
```
Same behaviour is observed on two different devices with the same setup (pi4, usb accel, py3.7)

mtyka, 2020-01-09:
Thank! Will take a look at this

mtyka, 2020-01-10:
I can confirm the leak, i see it ticking up also. We'll debug and update. Thank you !

Work around for now is to revert this repo to 
https://github.com/google-coral/examples-camera/commit/ae9e9df3398853ac39e45eaa629b5e9f0617711d

I.e. before the port to the new tflite API. It will use the older "edgetpu" API  instead but the feature set will be identical, so you're not loosing anything.

oliver9523, 2020-05-01:
any progress on this issue?

mtyka, 2020-05-02:
Sorry - I forgot to update this bug. Yes, I believe this was fixed in the latest release.

closed 2020-05-02 by mtyka