google-coral/examples-camera #32: vidioc_streamon no space left on device

12345k, 2020-01-14:
When I try to run the model by using  two webcams from USB hub I get this issue

vidioc_streamon no space left on device
VIDEOIO ERROR: V4L2: Could not obtain specifics of the capture window.
VIDEOIO ERROR: V4L: can't open camera by index 1
/dev/video1 does not support memory mapping
<VideoCapture 0x7f21d87e7ed0>
Traceback (most recent call last):
  File "test.py", line 13, in <module>
    gray = cv2.cvtColor(frame, cv2.COLOR_BGR2GRAY)
cv2.error: OpenCV(4.1.1) /io/opencv/modules/imgproc/src/color.cpp:182: error: (-215:Assertion failed) !_src.empty() in function 'cvtColor'

Can you please suggest me a way to run multiple cameras feeds on google dev board

Namburger, 2020-02-06:
@12345k seems like you are not the only one experiencing this issue, can you try out the the script I written [here](https://github.com/google-coral/edgetpu/issues/50#issuecomment-582971020) to see if the issue is still reproducible?

mbrooksx, 2020-02-07:
Hi @12345k 

Here's a minimal gstreamer example: https://gist.github.com/mbrooksx/2bb6dfa1bdc74512688d7ffd588dc832

Note you need to install gstreamer1.0-tools

I have two versions in the script, one that takes raw frames at VGA resolution and one that has higher res which I then decode. I'm not sure how you're receiving frames in your version but I wouldn't expect hires raw frames to work.

Thanks,
Michael

manoj7410, 2021-07-01:
@12345k Feel free to reopen if this is still an issue.

closed 2021-07-01 by manoj7410