google-coral/examples-camera #33: No matching distribution found for opencv-contrib-python==4.1.0.25

12345k, 2020-01-20:
When I was installing opencv requirment I get this issue 

mendel@bored-horse:~/latest_github/examples-camera/opencv$ bash install_requirem
ents.sh 
ERROR: Could not find a version that satisfies the requirement opencv-contrib-py
thon==4.1.0.25 (from versions: none)
ERROR: No matching distribution found for opencv-contrib-python==4.1.0.25
Reading package lists... Done
Building dependency tree       
Reading state information... Done
Note, selecting 'libhdf5-100' for glob 'libhdf5-1*'
Note, selecting 'libhdf5-103' for glob 'libhdf5-1*'
E: Unable to locate package libjasper1
Reading package lists... Done
Building dependency tree       
Reading state information... Done
python3-opencv is already the newest version (3.2.0+dfsg-6).
0 upgraded, 0 newly installed, 0 to remove and 1 not upgraded.

Any help on this issue?

mtyka, 2020-01-20:
Try opencv-contrib-python>=4.0,<4.1.0.25

12345k, 2020-01-20:
Same error message I get

mtyka, 2020-01-20:
Are you using the latest Mendel release (Mendel Day) ? Make sure your board
is flashed to the latest image.

12345k, 2020-01-21:
I am using mendel 4.0

mtyka, 2020-01-22:
Ok, apologies about that - i fixed the issue (I think) in https://github.com/google-coral/examples-camera/commit/8142340b1743269eb5bc6b01a1caf3fe06d77eb6
That package was only supposed to be installed for Raspbian. On Mendel the python3-opencv package should be sufficient.

12345k, 2020-01-22:
okay, thanks for response

closed 2020-01-22 by 12345k