google-coral/examples-camera #39: RuntimeError Raspberry Pi

MichielBbal, 2020-05-14:
After installation on my Raspberry Pi 4 I get the following error:
pi@raspberrypi:~/google-coral/examples-camera/opencv $ python3 detect.py
Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
Traceback (most recent call last):
  File "detect.py", line 131, in <module>
    main()
  File "detect.py", line 91, in main
    interpreter.allocate_tensors()
  File "/usr/local/lib/python3.7/dist-packages/tflite_runtime/interpreter.py", line 242, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/usr/local/lib/python3.7/dist-packages/tflite_runtime/interpreter_wrapper.py", line 115, in AllocateTensors
    return _interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
RuntimeError: Internal: Unsupported data type in custom op handler: 0Node number 2 (EdgeTpuDelegateForCustomOp) failed to prepare.

Namburger, 2020-06-12:
Can you share this command?
```
spkg -l | grep libedgetpu
```

rcaudill, 2020-12-10:
I'm getting the same error on the Pi 4. Here is my output from the requested command:

> pi@raspberrypi:~/coral/examples-camera/raspicam $ dpkg -l | grep libedgetpu
ii  libedgetpu1-legacy-std:armhf         15.0                                armhf        Support library for Edge TPU

Namburger, 2020-12-10:
@rcaudill please also make sure to upgrade your [tflite_runtime](https://www.tensorflow.org/lite/guide/python#install_just_the_tensorflow_lite_interpreter) package. it should be version 2.5.0 now!

rcaudill, 2020-12-11:
@Namburger thanks for the quick response. It looks like 2.5 is installed. Does this look correct:

> $ python3 -c 'print(__import__("tflite_runtime").__version__)'
2.5.0

Namburger, 2020-12-11:
@rcaudill ahhh, no problems! 
That looks okay, although I just realized that your libedgetpu is a legacy package (libedgetpu1-legacy-std:armhf 15.0), you should upgrade it!
```
$ sudo apt purge libedgetpu1-legacy-std
$ sudo apt install python3-pycoral
```
Verify that yo no longer have the legacy package (should looks like this now):
```
$ dpkg -l | grep edgetpu
ii  libedgetpu1-std:arm64                15.0                                arm64        Support library for Edge TPU
```

Lastly, sometimes linux wouldn't allows a certain usb rules by default so you may need this also:
```
$ sudo usermod -aG plugdev $USER
$ suo shutdown now
```

rcaudill, 2020-12-16:
@Namburger updating libedgetpu fixed the problem. Thanks!

closed 2021-07-01 by manoj7410