google-coral/examples-camera #41: Segmentation Fault using gstreamer for both classify.py and detect.py

JSwaim22, 2020-05-15:
I am using a coral dev board and a coral camera. I followed the setup instructions and then ran python3 classify.py and python3 detect.py. Both ended up giving me a list of failures ending in a segmentation fault. I tried running them both while connected to a monitor and when not (I wasn't entirely clear on where the camera feed was supposed to show up).

I included pictures below. Any help would be greatly appreciated!

Thanks!

[image]
[image]
[image]

JSwaim22, 2020-05-18:
issue resolved

closed 2020-05-18 by JSwaim22

alexanderfrey, 2020-09-01:
Having similar issue. Did you attach a monitor ? I thought the demo runs via stream...
```
python3 /home/mendel/google-coral/examples-camera/gstreamer/classify.py --videosrc /dev/video1
Unable to init server: Could not connect: Connection refused

(classify.py:6429): Gtk-WARNING **: 10:52:41.904: Locale not supported by C library.
	Using the fallback 'C' locale.
Unable to init server: Could not connect: Connection refused
Loading ../all_models/mobilenet_v2_1.0_224_quant_edgetpu.tflite with ../all_models/imagenet_labels.txt labels.
Detected Edge TPU dev board.
Gstreamer pipeline:
 v4l2src device=/dev/video1 ! video/x-raw,width=640,height=480,framerate=30/1 ! decodebin ! glupload ! tee name=t
            t. ! queue ! glfilterbin filter=glbox name=glbox ! video/x-raw,format=RGB,width=224,height=224 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
            t. ! queue ! glsvgoverlaysink name=overlaysink
        

(classify.py:6429): Gtk-CRITICAL **: 10:52:42.636: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(classify.py:6429): Gtk-CRITICAL **: 10:52:42.636: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(classify.py:6429): Gtk-CRITICAL **: 10:52:42.636: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
Segmentation fault
```