google-coral/examples-camera #45: What would be the best way to have the gstreamer example run without being connected to a montitor?

JSwaim22, 2020-06-11:
I want to run the gstreamer example on my Coral Dev Board but I need it to be able to work even when not connected to a monitor. Are there a couple of lines of code I can change to make this work?

mtyka, 2020-06-11:
you have to change the screen sink ("e.g. ximagesink") to "fakesink".
In the case of the devboard youd have to replace glsvgoverlaysink with "fakesink". However that will also remove the reference to the svg overlay so you will have to comment out any references to that too. 
I'm not aware of a good way to autodetect the monitor though.

closed 2020-06-11 by JSwaim22

Guillaume-slize, 2020-12-20:
Hey @JSwaim22,

I am also implementing a headless setup. I managed to do it with the previous architecture where you had a display option using this [thread](https://github.com/googlecodelabs/edgetpu-classifier/issues/1#issuecomment-509784840).
Did you end up implementing mtyka's answer?
@mtyka I have issues understanding exactly what your answer entails, could you develop it a bit?

Guillaume-slize, 2020-12-21:
After fiddling around with the different examples, the easiest way I found to enable headless object detection was to use the open CV script and comment this [line](https://github.com/google-coral/examples-camera/blob/c74a38362b05c07dd62e2e7c77d04a72d2164497/opencv/detect.py#L110).
At least it works for the application I wanted to build.