google-coral/examples-camera #52: Overlaysink error

320064714, 2020-08-12:
Getting the following error message when executing _classify.py_ on the Coral Board using the Coral Camera:

```
mendel@queen-tang:~$ sudo python3 /home/mendel/google-coral/examples-camera/gstreamer/classify.py  --model /home/mendel/google-coral/examples-camera/all_models/inception_v2_224_quant_edgetpu.tflite --labels /home/mendel/google-coral/examples-camera/all_models/imagenet_labels.txt

Loading /home/mendel/google-coral/examples-camera/all_models/inception_v2_224_quant_edgetpu.tflite with /home/mendel/google-coral/examples-camera/all_models/imagenet_labels.txt labels.
Detected Edge TPU dev board.
Gstreamer pipeline:
 v4l2src device=/dev/video0 ! video/x-raw,width=640,height=480,framerate=30/1 ! decodebin ! glupload ! tee name=t
            t. ! queue ! glfilterbin filter=glbox name=glbox ! video/x-raw,format=RGB,width=224,height=224 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
            t. ! queue ! glsvgoverlaysink name=overlaysink
        

(classify.py:5836): Gdk-CRITICAL **: 08:03:50.953: gdk_wayland_window_get_wl_surface: assertion 'GDK_IS_WAYLAND_WINDOW (window)' failed
Traceback (most recent call last):
  File "/home/mendel/google-coral/examples-camera/gstreamer/classify.py", line 115, in <module>
    main()
  File "/home/mendel/google-coral/examples-camera/gstreamer/classify.py", line 112, in main
    videofmt=args.videofmt)
  File "/home/mendel/google-coral/examples-camera/gstreamer/gstreamer.py", line 257, in run_pipeline
    pipeline = GstPipeline(pipeline, user_function, src_size)
  File "/home/mendel/google-coral/examples-camera/gstreamer/gstreamer.py", line 50, in __init__
    self.setup_window()
  File "/home/mendel/google-coral/examples-camera/gstreamer/gstreamer.py", line 168, in setup_window
    self.overlaysink.set_window_handle(wl_handle)
TypeError: Argument 1 does not allow None as a value

```

alexanderfrey, 2020-09-01:
I'm also getting error on dev board connected via ssh:
`python3 detect.py --videosrc /dev/video1 --videofmt jpeg
Unable to init server: Could not connect: Connection refused

(detect.py:6392): Gtk-WARNING **: 10:46:21.373: Locale not supported by C library.
	Using the fallback 'C' locale.
Unable to init server: Could not connect: Connection refused
Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
Detected Edge TPU dev board.
Gstreamer pipeline:
 v4l2src device=/dev/video1 ! image/jpeg,width=640,height=480,framerate=30/1 ! decodebin ! glupload ! tee name=t
            t. ! queue ! glfilterbin filter=glbox name=glbox ! video/x-raw,format=RGB,width=300,height=300 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
            t. ! queue ! glsvgoverlaysink name=overlaysink
        

(detect.py:6392): Gtk-CRITICAL **: 10:46:22.095: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(detect.py:6392): Gtk-CRITICAL **: 10:46:22.095: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(detect.py:6392): Gtk-CRITICAL **: 10:46:22.095: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
Segmentation fault`

dcohen-hub, 2021-02-02:
I get that as well. If connected via HDMI to monitor it resolves.
But I want to run it without HDMI connected