google-coral/examples-camera #54: Raspberry Pi AttributeError: 'Image' object has no attribute 'read'

FunkyGandalf, 2020-08-20:
Hello, I am having difficulty with this error and can't seem to diagnose it.  I am a newcomer to this, so any help would be greatly appreciated.

Goal:  I have used Teachable machine to create a squirrel/raccoon/bird detector.  The model was trained and working well on the  website.  I exported the model to Tensorflow Lite Edge TPU.  I copied the code snipped into a file called edgetpu_model.py as below.  I followed all the instructions to install the dependencies, openCV, and the tensorflow lite EdgeTPU library and have gone through those instructions several times to make sure everything was installed correctly, as far as I can tell it's all there.

When I run the program, it generates this error log:

Traceback (most recent call last):
  File "EdgeTPUModel.py", line 61, in <module>
    main()
  File "EdgeTPUModel.py", line 51, in main
    results = classifyImage(pil_im, engine)
  File "EdgeTPUModel.py", line 24, in classifyImage
    image = Image.open(image_path)
  File "/usr/lib/python3/dist-packages/PIL/Image.py", line 2643, in open
    prefix = fp.read(16)
AttributeError: 'Image' object has no attribute 'read'

------------------
(program exited with code: 1)
Press return to continue

Thanks for any help in getting this working!

**Here's the program code for Python Script edge_tpu.py:**

from edgetpu.classification.engine import ClassificationEngine
from PIL import Image
import cv2
import re
import os

# the TFLite converted to be used with edgetpu
modelPath = 'model_edgetpu.tflite'

# The path to labels.txt that was downloaded with your model
labelPath = 'labels.txt'

# This function parses the labels.txt and puts it in a python dictionary
def loadLabels(labelPath):
    p = re.compile(r'\s*(\d+)(.+)')
    with open(labelPath, 'r', encoding='utf-8') as labelFile:
        lines = (p.match(line).groups() for line in labelFile.readlines())
        return {int(num): text.strip() for num, text in lines}

# This function takes in a PIL Image from any source or path you choose
def classifyImage(image_path, engine):
    # Load and format your image for use with TM2 model
    # image is reformated to a square to match training
    image = Image.open(image_path)
    image.resize((224, 224))

    # Classify and ouptut inference
    classifications = engine.ClassifyWithImage(image)
    return classifications

def main():
    # Load your model onto your Coral Edgetpu
    engine = ClassificationEngine(modelPath)
    labels = loadLabels(labelPath)

    cap = cv2.VideoCapture(0)
    while cap.isOpened():
        ret, frame = cap.read()
        if not ret:
            break

        # Format the image into a PIL Image so its compatable with Edge TPU
        cv2_im = frame
        pil_im = Image.fromarray(cv2_im)

        # Resize and flip image so its a square and matches training
        pil_im.resize((224, 224))
        pil_im.transpose(Image.FLIP_LEFT_RIGHT)

        # Classify and display image
        results = classifyImage(pil_im, engine)
        cv2.imshow('frame', cv2_im)
        print(results)
        if cv2.waitKey(1) & 0xFF == ord('q'):
            break

    cap.release()
    cv2.destroyAllWindows()

if __name__ == '__main__':
    main()

Namburger, 2020-10-27:
hello, it's really hard to say without at least seeing the implementation of of `results = classifyImage(pil_im, engine)`. Is it from this repo? I'm only asking because I'm not familiar with *EdgeTPUModel.py*.
Anyhow, the error that you are seeing is a PIL library issue when loading the image

bricklane, 2020-11-24:
I'm having a very similar issue on the coral dev board.

QuietlyAmusing, 2021-03-02:
Hello, I think I have found a **solution** to this problem.

(I have also used Google's Teachable Machine to create my own image classification model and exported it for use on a Raspberry Pi with a Coral USB accelerator. I copied the code from the Teachable Machine website and only made modifications to the `modelPath` and `labelPath`.)

I had the same error as above and this is what I did to resolve the issue:

For line 24:
`image = Image.open(image_path)`

change it to:
`image = image_path`

and the program should run without the issue.

Hope this helps!

QuietlyAmusing, 2021-03-02:
Alternatively, you could modify the `classifyImage()` function from this:

```
def classifyImage(image_path, engine):
    # Load and format your image for use with TM2 model
    # image is reformated to a square to match training
    image = Image.open(image_path)
    image.resize((224, 224))

    # Classify and ouptut inference
    classifications = engine.ClassifyWithImage(image)
    return classifications
```

to the following:

```
def classifyImage(image, engine):
    # Load and format your image for use with TM2 model
    # image is reformated to a square to match training
    image.resize((224, 224))`

    # Classify and ouptut inference
    classifications = engine.ClassifyWithImage(image)
    return classifications
```

P.S. I am new to GitHub so it would be helpful if a more experienced user "moves" this solution to the appropriate place.

closed 2023-06-27 by hjonnala