google-coral/examples-camera #58: gstreamer rtsp error?

ymk201, 2020-11-10:
INPUT
 $ python3 classify.py --videosrc 'rtsp-url' // have checked with vlc and the 'rtsp-url' works fine.

OUTPUT
Loading ../all_models/mobilenet_v2_1.0_224_quant_edgetpu.tflite with ../all_models/imagenet_labels.txt labels.
Gstreamer pipeline:
 rtspsrc location='rtsp-url' ! tee name=t
            t. ! queue max-size-buffers=1 leaky=downstream ! videoconvert ! videoscale ! video/x-raw,width=224,height=168 ! videobox name=box autocrop=true
               ! video/x-raw,format=RGB,width=224,height=224 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
            t. ! queue max-size-buffers=1 leaky=downstream ! videoconvert
               ! rsvgoverlay name=overlay ! videoconvert ! ximagesink sync=false
            
Warning: gst_parse_error: Delayed linking failed. (7): ./grammar.y(506): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstTee named t

Namburger, 2020-11-10:
@ymk201 Hey!
This seems to be a gstreamer error or could be a problem with your rtsp-url, maybe you can try to test your `rtsp-url` on a gst-launch-1.0 first?
```
gst-launch-1.0 rtspsrc location=
```

ymk201, 2020-11-11:
it seems that my gst were broken some how. I'm still working on it.
It works fine few weeks ago.

My gstreamer stops with bellow error, is there any suggest for me?
I have already tried reinstall the gst. but still getting same error.
(it works fine with fakesink.)
ERROR: from element /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc4: Internal data stream error.
Additional debug info:
gstbasesrc.c(3072): gst_base_src_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc4:
streaming stopped, reason not-linked (-1)

PortScanner, 2021-07-19:
Have you found a solution?

closed 2021-09-23 by ymk201

jiekechoo, 2022-07-28:
It wors for me :

python3 detect.py --videosrc "rtsp://192.168.120.58/ch01_sub.h264 ! rtph264depay ! h264parse ! avdec_h264 ! videoconvert"