google-coral/examples-camera #59: Unknown pixelformat

bricklane, 2020-11-24:
Trying to run detect.py on the coral dev board, using coral camera, and running into this issue continually.

`$ python3 detect.py`

`Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.`

`[ 2440.991079] unknown pixelformat:'BGR3'`
`[ 2441.036095] mx6s-csi 30a90000.csi1_bridge: Fourcc format (0x33524742) invalid.`

`[ 2441.122862] unknown pixelformat:'YV12'`
`[ 2441.167939] mx6s-csi 30a90000.csi1_bridge: Fourcc format (0x32315659) invalid.`

`[ 2441.254807] unknown pixelformat:'411P'`
`[ 2441.299827] mx6s-csi 30a90000.csi1_bridge: Fourcc format (0x50313134) invalid.`

`[ 2441.386643] unknown pixelformat:'MJPG'`
`[ 2441.431718] mx6s-csi 30a90000.csi1_bridge: Fourcc format (0x47504a4d) invalid.`

`[ 2441.518516] unknown pixelformat:'JPEG'`
`[ 2441.563610] mx6s-csi 30a90000.csi1_bridge: Fourcc format (0x4745504a) invalid.`

`Unable to init server: Could not connect: Connection refused`

`(frame:4255): Gtk-WARNING **: 23:38:18.439: cannot open display:`

Namburger, 2020-11-25:
@bricklane it is actually defaulted to [/dev/video0](https://github.com/google-coral/examples-camera/blob/master/gstreamer/detect.py#L119).
Although, this repo is being updated currently and won't works since our edgetpu API is deprecated for pycoral. We're actively working on updating this and will have it working soon. Apologies!

bricklane, 2020-11-27:
Thanks for the info! To clarify, are you saying that using a usb camera and switching to /dev/video1 would be a solve in the meantime?

Namburger, 2020-11-27:
Hello, no, unfortunately the issue is with the edgetpu API (the AI inference library) and not the video inputs :/

Namburger, 2020-11-27:
Here is a quick detection srcipt if you want to use this in the meanwhile: https://gist.github.com/Namburger/f7e6c18af94ef3d6a70076a130eb1f7c

Namburger, 2020-11-27:
But maybe you can use this detection script in the mean while?
https://gist.github.com/Namburger/f7e6c18af94ef3d6a70076a130eb1f7c

bricklane, 2020-11-28:
That makes sense. I'll look into it - thanks!

manoj7410, 2021-07-05:
Feel free to reopen if the issue still persists.

closed 2021-07-05 by manoj7410