google-coral/examples-camera #60: Unable to Init Server - Coral Dev Board / Coral Camera - HELPPPP!!!

bettencb, 2020-11-25:
# Coral Camera Error on Coral Dev Board
* *                                                  * *
> **Note**: I'm connected through OTG USB connection and the mdt shell on Ubuntu 20.10. Neither classify or detect python scripts run. Any help would be much appreciated as I've encountered many similar errors with other samples but have successfully ran then image classification python script when using just a .jpg image.
```
mendel@wishful-bunny:~/google-coral/examples-camera/gstreamer$ sudo python3 classify.py 
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
Loading ../all_models/mobilenet_v2_1.0_224_quant_edgetpu.tflite with ../all_models/imagenet_labels.txt labels.
Traceback (most recent call last):
  File "classify.py", line 115, in <module>
    main()
  File "classify.py", line 82, in main
    interpreter.allocate_tensors()
  File "/usr/local/lib/python3.7/dist-packages/tflite_runtime/interpreter.py", line 242, in allocate_tensors
    return self._interpreter.AllocateTensors()
  File "/usr/local/lib/python3.7/dist-packages/tflite_runtime/interpreter_wrapper.py", line 115, in AllocateTensors
    return _interpreter_wrapper.InterpreterWrapper_AllocateTensors(self)
RuntimeError: Internal: Unsupported data type in custom op handler: 0Node number 1 (EdgeTpuDelegateForCustomOp) failed to prepare.
```

Namburger, 2020-11-25:
Ahh, sorry, this repo is being updated currently and won't works since our edgetpu API is deprecated for pycoral. We're actively working on updating this and will have it working soon!

bettencb, 2020-11-27:
First off happy thanksgiving, secondly, could you possibly let me know through this issue log when the repo is up and running again @Namburger

Namburger, 2020-11-27:
Hello, yes I will, please expect a few weeks, for now i do have this detection script that maybe you can use?
https://gist.github.com/Namburger/f7e6c18af94ef3d6a70076a130eb1f7c

closed 2020-12-07 by bettencb