google-coral/examples-camera #61: Compatible with pycoral and eagle enterprise?

startupstarter, 2020-12-07:
oading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
Detected Edge TPU dev board.
Gstreamer pipeline:
 v4l2src device=/dev/video0 ! video/x-raw,width=640,height=480,framerate=30/1 ! decodebin ! glupload ! tee name=t
            t. ! queue ! glfilterbin filter=glbox name=glbox ! video/x-raw,format=RGB,width=300,height=300 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
            t. ! queue ! glsvgoverlaysink name=overlaysink
        

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.707: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.711: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.712: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.712: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.712: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.712: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.713: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.713: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.713: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.714: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.714: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.714: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.714: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.714: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.714: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.714: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.714: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.715: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.715: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.715: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.715: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.715: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.715: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.715: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.715: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.716: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.716: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.716: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.717: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.717: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.717: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

[... 69 lines cut ...]

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.830: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.830: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.830: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.830: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.830: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.830: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.830: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.830: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.830: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.831: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.831: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed

(detect.py:4010): Gdk-CRITICAL **: 04:35:48.831: gdk_monitor_get_scale_factor: assertion 'GDK_IS_MONITOR (monitor)' failed
Segmentation fault

Namburger, 2020-12-07:
Hi, this is because this API is deprecated and we are working to get this updated, please check here for update:
https://github.com/google-coral/examples-camera/issues/60#issuecomment-733732762

closed 2020-12-07 by Namburger

startupstarter, 2020-12-07:
> Hi, this is because this API is deprecated and we are working to get this updated, please check here for update:
> [#60 (comment)](https://github.com/google-coral/examples-camera/issues/60#issuecomment-733732762)

With the new posenet commit it now works. I guess it was related to gstreamer. The segment fault issue was related to not having the HDMI output connected.