google-coral/examples-camera #62: Error when using inception_v1_224_quant_edgetpu.tflite in OpenCV demo.

arrowkato, 2020-12-10:
I use a MacBook Pro (OS 10.15) and a Coral.
I executed this command and it works fine.
```
python3 detect.py \
  --model ../all_models/mobilenet_ssd_v2_face_quant_postprocess_edgetpu.tflite \
  --labels ../all_models/coco_labels.txt
```

However, I executed below command and it showed errors.

```
python3 detect.py \
  --model ../all_models/inception_v1_224_quant_edgetpu.tflite \
  --labels ../all_models/imagenet_labels.txt
```

```
Loading ../all_models/inception_v1_224_quant_edgetpu.tflite with ../all_models/imagenet_labels.txt labels.
Traceback (most recent call last):
  File "detect.py", line 131, in <module>
    main()
  File "detect.py", line 107, in main
    objs = get_output(interpreter, score_threshold=args.threshold, top_k=args.top_k)
  File "detect.py", line 57, in get_output
    class_ids = common.output_tensor(interpreter, 1)
  File "/Users/shinji/vscode_projects/arrowkato_repo/Coral_Edge_TPU/example-camera/opencv/common.py", line 53, in output_tensor
    output_details = interpreter.get_output_details()[i]
IndexError: list index out of range
```

Is there anything additional I need to set up?

Namburger, 2020-12-10:
Hello, that will not work as the second model you are using is a classification model as opposed to a detection model!

arrowkato, 2020-12-11:
Thank you for your replying.

I understood it.

closed 2020-12-11 by arrowkato