google-coral/examples-camera #63: gstreamer: Could not initialise X output (10)

tamberg, 2020-12-29:
Hi, trying to run _examples-camera/gstreamer_ on the Coral Dev Board Mini (with a HDMI display connected) I get:
```
$ bash install_requirements.sh
$  python3 detect.py 
(detect.py:1035): dconf-WARNING **: 10:45:24.153: Unable to open /usr/local/share/dconf/profile/user: Not a directory
Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
Gstreamer pipeline:
 v4l2src device=/dev/video0 ! video/x-raw,width=640,height=480,framerate=30/1 ! tee name=t
            t. ! queue max-size-buffers=1 leaky=downstream ! videoconvert ! videoscale ! video/x-raw,width=300,height=225 ! videobox name=box autocrop=true
               ! video/x-raw,format=RGB,width=300,height=300 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
            t. ! queue max-size-buffers=1 leaky=downstream ! videoconvert
               ! rsvgoverlay name=overlay ! videoconvert ! ximagesink sync=false
            
Error: gst-resource-error-quark: Could not initialise X output (10): ximagesink.c(860): gst_x_image_sink_xcontext_get (): /GstPipeline:pipeline0/GstXImageSink:ximagesink0:
Could not open display   
```
Any ideas how to fix this? (I'm not a Linux expert, btw.)

Kind regards, Thomas

hjonnala, 2021-08-09:
@tamberg are you still facing the issue?

closed 2022-02-09 by hjonnala

thedevleon, 2022-04-04:
I'm having the same issue.

Relevant output:
```
mendel@jumping-rabbit:~/coral/example-object-tracker/gstreamer$ python3 detect.py 

(detect.py:2050): dconf-WARNING **: 16:29:45.251: Unable to open /usr/local/share/dconf/profile/user: Not a directory
Loading ../models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../models/coco_labels.txt labels.
Gstreamer pipeline:
 v4l2src device=/dev/video0 ! video/x-raw,width=640,height=480,framerate=30/1 ! tee name=t
            t. ! queue max-size-buffers=1 leaky=downstream ! videoconvert ! videoscale ! video/x-raw,width=300,height=225 ! videobox name=box autocrop=true
               ! video/x-raw,format=RGB,width=300,height=300 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
            t. ! queue max-size-buffers=1 leaky=downstream ! videoconvert
               ! rsvgoverlay name=overlay ! videoconvert ! ximagesink sync=false
            
Traceback (most recent call last):
  File "detect.py", line 218, in <module>
    main()
  File "detect.py", line 214, in main
    videofmt=args.videofmt)
  File "/home/mendel/coral/example-object-tracker/gstreamer/gstreamer.py", line 275, in run_pipeline
    pipeline = GstPipeline(pipeline, user_function, src_size, mot_tracker)
  File "/home/mendel/coral/example-object-tracker/gstreamer/gstreamer.py", line 39, in __init__
    self.pipeline = Gst.parse_launch(pipeline)
gi.repository.GLib.Error: gst_parse_error: no element "ximagesink" (1)

```

hjonnala, 2022-04-04:
@thedevleon seems to be you are working with [example-object-tracker](https://github.com/google-coral/example-object-tracker/tree/master/gstreamer) repo. example-object-tracker repo has not been updated for dev board mini.

Thanks!

kadjacent, 2023-10-30:
any update on this? I am getting the same error.