google-coral/examples-camera #65: Building the nativeapp failed

ting3000, 2021-01-26:
I try to build the nativeapp on Coral Dev Board, it failed because the tensorflow library issue. First, it seems that the Makefile from tensorflow and nativeapp point to different path, second issue is the missing function from tensorflow library. Please help!

Here are the details.

I running build_aarch64_lib.sh, the library is created under
tensorflow/lite/tools/make/gen/**linux_aarch64**/lib/libtensorflow-lite.a

But the Makefile to build nativeapp has
TFLITE_OBJS := $(TENSORFLOW_DIR)/tensorflow/lite/tools/make/gen/**aarch64_armv8-a**/lib/libtensorflow-lite.a

The directory doesn't match.

I notice that the Makefile to build libtensorflow-lite.a has
TARGET_OUT_DIR ?= $(TARGET)_$(TARGET_ARCH)
And the build_aarch64_lib.sh has 
make -j ${NO_JOB} TARGET=aarch64 -C "${TENSORFLOW_DIR}" -f tensorflow/lite/tools/make/Makefile $@

How should I fix this problem?

I tried to fix the path of TFLITE_OBJS in Makefile of nativeapp, then I got this error from running make:
g++  -o "mendelcam" main.cc camerastreamer.cc inferencewrapper.cc /home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/tools/make/gen/linux_aarch64/lib/libtensorflow-lite.a -std=c++11 -O3 -Wall -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/home/mendel/sdcard/spacer/development/google-coral/tensorflow -I/home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/tools/make/downloads/flatbuffers/include  -lrt -ldl -lpthread -lgstreamer-1.0 -lglib-2.0 -lgobject-2.0 -ledgetpu
inferencewrapper.cc: In constructor ‘coral::InferenceWrapper::InferenceWrapper(const string&, const string&)’:
inferencewrapper.cc:58:74: error: no matching function for call to ‘tflite::Interpreter::ModifyGraphWithDelegate(<brace-enclosed initializer list>)’
   interpreter_->ModifyGraphWithDelegate({delegate, edgetpu_free_delegate});

manoj7410, 2021-07-01:
did you try a cross build on x86_64 machine instead of building on the dev board ?

ting3000, 2021-07-01:
The same problem happened to me when I tried to build on dev board or x86_64 linux box. The log I post here was from the build on coral dev board.

manoj7410, 2021-07-02:
@ting3000 Which is this native app? Is it an open source ? Are there any steps that you can share to reproduce this issue ?

ting3000, 2021-07-02:
I was following this document (https://github.com/google-coral/examples-camera/tree/master/nativeapp#build-and-install-libtensorflow-litea) to "Build and install libtensorflow-lite.a"

ting3000, 2021-07-02:
The nativeapp is the sample app in this repository: https://github.com/google-coral/examples-camera/tree/master/nativeapp

manoj7410, 2021-07-12:
Are there any particular steps to reproduce this issue ?