google-coral/examples-camera #66: Compilation of inferencewrapper.cc failed

ting3000, 2021-01-27:
Compilation of inferencewrapper.cc failed with the error message "inferencewrapper.cc:58:74: error: no matching function for call to ‘tflite::Interpreter::ModifyGraphWithDelegate(<brace-enclosed initializer list>)’"

The tensorflow source code was from "git clone https://github.com/tensorflow/tensorflow"

Could somebody help how to fix the problem? Thanks!

Here is the output of make command:
g++  -o "mendelcam" main.cc camerastreamer.cc inferencewrapper.cc /home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/tools/make/gen/linux_aarch64/lib/libtensorflow-lite.a -std=c++11 -O3 -Wall -I/usr/lib/aarch64-linux-gnu/glib-2.0/include -I/usr/include/gstreamer-1.0 -I/usr/include/glib-2.0 -I/home/mendel/sdcard/spacer/development/google-coral/tensorflow -I/home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/tools/make/downloads/flatbuffers/include  -lrt -ldl -lpthread -lgstreamer-1.0 -lglib-2.0 -lgobject-2.0 -ledgetpu
inferencewrapper.cc: In constructor ‘coral::InferenceWrapper::InferenceWrapper(const string&, const string&)’:
inferencewrapper.cc:58:74: error: no matching function for call to ‘tflite::Interpreter::ModifyGraphWithDelegate(<brace-enclosed initializer list>)’
   interpreter_->ModifyGraphWithDelegate({delegate, edgetpu_free_delegate});
                                                                          ^
In file included from inferencewrapper.h:9,
                 from inferencewrapper.cc:1:
/home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/interpreter.h:489:16: note: candidate: ‘TfLiteStatus tflite::Interpreter::ModifyGraphWithDelegate(TfLiteDelegate*)’
   TfLiteStatus ModifyGraphWithDelegate(TfLiteDelegate* delegate);
                ^~~~~~~~~~~~~~~~~~~~~~~
/home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/interpreter.h:489:16: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘TfLiteDelegate*’
/home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/interpreter.h:499:23: note: candidate: ‘template<class Delegate, class Deleter> TfLiteStatus tflite::Interpreter::ModifyGraphWithDelegate(std::unique_ptr<_Tp, _Dp>)’
   inline TfLiteStatus ModifyGraphWithDelegate(
                       ^~~~~~~~~~~~~~~~~~~~~~~
/home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/interpreter.h:499:23: note:   template argument deduction/substitution failed:
inferencewrapper.cc:58:74: note:   couldn't deduce template parameter ‘Delegate’
   interpreter_->ModifyGraphWithDelegate({delegate, edgetpu_free_delegate});
                                                                          ^
make: *** [Makefile:38: mendelcam] Error 1

Namburger, 2021-01-28:
Hello @ting3000 did you checkout the tensorflow commit as documented here: https://github.com/google-coral/examples-camera/tree/master/nativeapp#build-and-install-libtensorflow-litea
Without checking out that specific commit, the library will either not build, or breaks during runtime

ting3000, 2021-01-28:
Hi @Namburger thank you for helping me on it! You are correct that I missed to set the TENSORFLOW_COMMIT before checkout. 

After I fixed and run the make again, I got the following error, do you have any suggestion how to solve it? Thanks!
In file included from tensorflow/lite/tools/make/downloads/absl/absl/base/internal/exponential_biased.cc:15:
/home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/tools/make/downloads/absl/absl/base/internal/exponential_biased.h:24:1: error: ‘ABSL_NAMESPACE_BEGIN’ does not name a type
 ABSL_NAMESPACE_BEGIN
 ^~~~~~~~~~~~~~~~~~~~
/home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/tools/make/downloads/absl/absl/base/internal/exponential_biased.h:127:1: error: ‘ABSL_NAMESPACE_END’ does not name a type; did you mean ‘ABSL_BASE_PORT_H_’?
 ABSL_NAMESPACE_END
 ^~~~~~~~~~~~~~~~~~
 ABSL_BASE_PORT_H_
tensorflow/lite/tools/make/downloads/absl/absl/base/internal/exponential_biased.cc:28:1: error: ‘ABSL_NAMESPACE_BEGIN’ does not name a type
 ABSL_NAMESPACE_BEGIN
 ^~~~~~~~~~~~~~~~~~~~
tensorflow/lite/tools/make/downloads/absl/absl/base/internal/exponential_biased.cc:92:1: error: ‘ABSL_NAMESPACE_END’ does not name a type; did you mean ‘ABSL_BASE_PORT_H_’?
 ABSL_NAMESPACE_END
 ^~~~~~~~~~~~~~~~~~
 ABSL_BASE_PORT_H_

ting3000, 2021-01-28:
Hi @Namburger after delete the previous source code, and chekout and download the dependency again, the previous error is solved. Now I get different error, I will open a new issue for it. Thanks!

closed 2021-01-28 by ting3000