google-coral/examples-camera #67: build_aarch64_lib.sh failed because undefined reference from the densify.cc

ting3000, 2021-01-28:
When building the specific version of libtensorflow-lite.a, I checkout the version of  TENSORFLOW_COMMIT="d855adfc5a0195788bf5f92c3c7352e638aa1109", which is from https://github.com/google-coral/edgetpu/blob/master/WORKSPACE

build_aarch64_lib.sh reported the following error. Could somebody helps.

/usr/bin/ld: /home/mendel/sdcard/spacer/development/google-coral/tensorflow/tensorflow/lite/tools/make/gen/aarch64_armv8-a/lib/libtensorflow-lite.a(densify.o): in function `tflite::ops::builtin::densify::Eval(TfLiteContext*, TfLiteNode*)':
densify.cc:(.text+0x348): undefined reference to `tflite::optimize::sparsity::FormatConverter<signed char>::FormatConverter(std::vector<int, std::allocator<int> > const&, TfLiteSparsity const&)'
/usr/bin/ld: densify.cc:(.text+0x354): undefined reference to `tflite::optimize::sparsity::FormatConverter<signed char>::SparseToDense(signed char const*)'
/usr/bin/ld: densify.cc:(.text+0x6c0): undefined reference to `tflite::optimize::sparsity::FormatConverter<float>::FormatConverter(std::vector<int, std::allocator<int> > const&, TfLiteSparsity const&)'
/usr/bin/ld: densify.cc:(.text+0x6cc): undefined reference to `tflite::optimize::sparsity::FormatConverter<float>::SparseToDense(float const*)'
collect2: error: ld returned 1 exit status

ting3000, 2021-01-28:
Hi @Namburger I saw you helped the similar issue [before](https://github.com/google-coral/edgetpu/issues/78#issue-581334241). Is my issue caused by checking out the wrong version of the code? Could you please help!
I am not sure if this is the proper place for creating this issue because it seems related to edgeptu, but I am trying to build nativeapp here. Thanks!

Namburger, 2021-01-28:
@ting3000 oh man.. now that I'm looking at this, this example need a refresh in build instruction, so sorry, let me try to work this out and will check back with you. 
The TENSORFLOW_COMMIT should really be coming from `libcoral` instead of `edgetpu` now (since edgetpu is deprecated). Apologies, I'll try to get this updated this week. 
Maybe in the mean time, you can checkout [this repo](https://github.com/namburger/edgetpu-detection-camera) of mine for a C++ example? it is using opencv instead of gstreamer but overall shouldn't be much different

ting3000, 2021-01-28:
@Namburger Thank you for helping me!

ting3000, 2021-02-08:
@Namburger Good morning! Is there update for the fix? I would like to use gstreamer because other dependency. Thanks!

ting3000, 2021-02-15:
@Namburger Do you have update on this? Thanks!

jedichen121, 2021-03-14:
Had the same problem. Changing the TENSORFLOW_COMMIT to the following helped me build successfully. 

TENSORFLOW_COMMIT = "48c3bae94a8b324525b45f157d638dfd4e8c3be1"

https://github.com/google-coral/libcoral/blob/master/WORKSPACE

manoj7410, 2021-07-05:
Feel free to reopen if the issue still persists.

closed 2021-07-05 by manoj7410