google-coral/examples-camera #69: Raspberry PI 4 64 Bit OS beta, Coral USB accelerator, classify_capture.py libmmal.so error

hackebike, 2021-02-16:
I am testing Google Coral TPU usb accelerator on a Raspberry PI 4 64 Bit OS beta, Kernel 5.10.

I am getting this error when I run the classify_capture.py script:

 ```
python classify_capture.py 
Traceback (most recent call last):
  File "classify_capture.py", line 24, in <module>
    import picamera
  File "/usr/local/lib/python3.7/dist-packages/picamera/__init__.py", line 72, in <module>
    from picamera.exc import (
  File "/usr/local/lib/python3.7/dist-packages/picamera/exc.py", line 41, in <module>
    import picamera.mmal as mmal
  File "/usr/local/lib/python3.7/dist-packages/picamera/mmal.py", line 49, in <module>
    _lib = ct.CDLL('libmmal.so')
  File "/usr/lib/python3.7/ctypes/__init__.py", line 356, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libmmal.so: cannot open shared object file: No such file or directory
```
Am I suppose to install the library? From where?

Namburger, 2021-02-16:
@hackebike that's a library that the pi OS is missing LOL

hackebike, 2021-02-17:
I know, but they had the package and libs and then they took it out. I playing guessing game!
thx for the support, other examples worked.