google-coral/examples-camera #72: Can't exit demo

Tylersuard, 2021-02-27:
Hello.  I got the gstreamer classify.py demo working, and it's great!  The problem is, I can't close it now.  Ctrl-c does nothing, neither does ctrl alt delete.  The instructions say never to unplug the coral dev board before shutting it down... what should I do?

CurlyWurly-1, 2021-03-05:
If you're using an old 4x3 monitor attached to the R.PI, you may find the taskbar is still displayed at the top.
If so, you can access the terminal session, close it and this cancels the screen being displayed .

If however, you try using a HDMI widescreen monitor attached to the R.PI (like I have tried), you may find the taskbar has dissappeared completely and all you see isthe camera image, with no apparent access to the toolbar - so you can't access the terminal session to kill it :(  
If that is the case, you can get round this by using VLC on a desktop (e.g. win10 PC) with a 2nd monitor attached to the R.PI.
For example, 
Let's say you access the R.PI using VLC, and execute the python program in a terminal session (all done via VLC).  
Once the python program is running OK. the 2nd monitor will then show the camera screen full size but you will see that the camera screen is NOT seen in the VLC session.
This means that you can easily access the terminal session in VLC ! 

TBH, this works better for me anyway - My flow of working mean I use the desktop PC most of the time and use VLC / SSH to access my various R.Pi machines anyway

Tylersuard, 2021-03-05:
Thank you for the response! I was actually using this on a coral dev board

Sent from ProtonMail Mobile

manoj7410, 2021-07-12:
@Tylersuard Did you try to SSH into the board and tried to kill the process using 'sudo kill -9 <PID>' ?