google-coral/examples-camera #74: Coral camera doesn't work on fresh flash of Coral dev board

strubell, 2021-04-17:
I'm trying to figure out why I can't get the Coral camera to work on my dev board. I just tried re-flashing the dev board, but I'm getting the same error as before. The dev board can see the camera; when I run `v4l2-ctl --list-devices` I get the following output, which looks right:
```
i.MX6S_CSI (platform:30a90000.csi1_bridge):
        /dev/video0
```
but when I try to take a test picture with `snapshot` I get the following warning, then it hangs:
```
(gst-plugin-scanner:4650): GStreamer-WARNING **: 15:34:54.752: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstchromaprint.so': /usr/lib/aarch64-linux-gnu/libgomp.so.1: cannot allocate memory in static TLS block
```
I also get a blank screen when I try to run the demos. It seems like a gstreamer library issue. Any pointers on how to go about debugging? Thanks!

strubell, 2021-04-17:
If I run `export LD_PRELOAD=/usr/lib/aarch64-linux-gnu/libgomp.so.1` before `snapshot` it stops producing that error, but doesn't solve the problem (`snapshot` still hangs and does not take a picture).

manoj7410, 2021-05-31:
@strubell What is the power supply that you are using ?

strubell, 2021-05-31:
I've tried two different fast-charging usb-c power banks. One is the Jackery Explorer 160 Portable Power Station which shows the Coral is drawing 3 watts when plugged in.

manoj7410, 2021-05-31:
Minimum requirement is 2Amp 5V, Is your adaptor/power bank meeting these requirements?

strubell, 2021-05-31:
Yes, it provides 3A/5V.

strubell, 2021-06-01:
It looks like an issue with gstreamer plugins so I tried running: 
`sudo apt-get install --reinstall gstreamer-1.0 gstreamer1.0-plugins-bad gstreamer1.0-plugins-base gstreamer1.0-plugins-good gstreamer1.0-plugins-ugly`

This seems to have gotten somewhere, as that install hangs on what I assume is a custom dpkg trigger related to the snapshot command: `Processing triggers fsnapshot`

strubell, 2021-06-03:
I ordered a new camera, and it works. The hardware was defective.

closed 2021-06-03 by strubell