google-coral/examples-camera #76: Coral dev mini opencv error

SunBeenMoon, 2021-05-24:
Hi, 
I'm testing the examples with coral mini board!

But, I've got the error. 
```
I can't execute the opencv examples. 
mendel@silly-quill:~/google-coral$ cd examples-camera/
mendel@silly-quill:~/google-coral/examples-camera$ ls
all_models       download_models.sh  LICENSE    opencv  raspicam
CONTRIBUTING.md  gstreamer           nativeapp  pygame  README.md
mendel@silly-quill:~/google-coral/examples-camera$ cd opencv/
mendel@silly-quill:~/google-coral/examples-camera/opencv$ ls
detect.py  install_requirements.sh  README.md
mendel@silly-quill:~/google-coral/examples-camera/opencv$ bash install_requirements.sh
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-opencv is already the newest version (3.2.0+dfsg-6).
0 upgraded, 0 newly installed, 0 to remove and 89 not upgraded.
mendel@silly-quill:~/google-coral/examples-camera/opencv$ python3 detect.py
Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
Unable to stop the stream: Invalid argument
https://pygments.org/demo/?lexer=python&code=mendel%40silly-quill%3A~%2Fgoogle-coral%24%20cd%20examples-camera%2F%0Amendel%40silly-quill%3A~%2Fgoogle-coral%2Fexamples-camera%24%20ls%0Aall_models%20%20%20%20%20%20%20download_models.sh%20%20LICENSE%20%20%20%20opencv%20%20raspicam%0ACONTRIBUTING.md%20%20gstreamer%20%20%20%20%20%20%20%20%20%20%20nativeapp%20%20pygame%20%20README.md%0Amendel%40silly-quill%3A~%2Fgoogle-coral%2Fexamples-camera%24%20cd%20opencv%2F%0Amendel%40silly-quill%3A~%2Fgoogle-coral%2Fexamples-camera%2Fopencv%24%20ls%0Adetect.py%20%20install_requirements.sh%20%20README.md%0Amendel%40silly-quill%3A~%2Fgoogle-coral%2Fexamples-camera%2Fopencv%24%20bash%20install_requirements.sh%0AReading%20package%20lists...%20Done%0ABuilding%20dependency%20tree%0AReading%20state%20information...%20Done%0Apython3-opencv%20is%20already%20the%20newest%20version%20(3.2.0%2Bdfsg-6).%0A0%20upgraded%2C%200%20newly%20installed%2C%200%20to%20remove%20and%2089%20not%20upgraded.%0Amendel%40silly-quill%3A~%2Fgoogle-coral%2Fexamples-camera%2Fopencv%24%20python3%20detect.py%0ALoading%20..%2Fall_models%2Fmobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite%20with%20..%2Fall_models%2Fcoco_labels.txt%20labels.%0AUnable%20to%20stop%20the%20stream%3A%20Invalid%20argument
```
 
The error message is so simple that I cant' figure out what the problem come from. 
I need help.

Regards,

manoj7410, 2021-05-28:
@SunBeenMoon Can you try to put print statements before and after https://github.com/google-coral/examples-camera/blob/master/opencv/detect.py#L61 ?  
Is there a display connected to your board ? 
I don't have a Dev Board Mini with me right now to try with.

SunBeenMoon, 2021-05-28:
Thank you for your reply @manoj7410 

I put 'Hi!' after the code. 

The output is below.
[image]

manoj7410, 2021-05-31:
@SunBeenMoon Can you try to run `sudo pip3 install opencv-contrib-python==4.1.0.25` and then try the demo again ?

SunBeenMoon, 2021-06-02:
Thank you for your reply @manoj7410 
However, opencv is not working.
[image]

manoj7410, 2021-06-02:
Can you try with `sudo pip3 install opencv-contrib-python==4.5.2.52` ?

SunBeenMoon, 2021-06-02:
It's not work...
[image]

```
mendel@silly-quill:~/google-coral/examples-camera/opencv$ bash install_requirements.sh
Reading package lists... Done
Building dependency tree
Reading state information... Done
python3-opencv is already the newest version (3.2.0+dfsg-6).
0 upgraded, 0 newly installed, 0 to remove and 89 not upgraded.
mendel@silly-quill:~/google-coral/examples-camera/opencv$ python3 detect.py
Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
Unable to stop the stream: Invalid argument
mendel@silly-quill:~/google-coral/examples-camera/opencv$ sudo pip3 install opencv-contrib-python==4.5.2.52
Collecting opencv-contrib-python==4.5.2.52
  Downloading https://files.pythonhosted.org/packages/c7/dc/0e9aedf4d26a4608ef1cd22a6083d88c1bc9207a8b07b6fc86a9f376e267/opencv-contrib-python-4.5.2.52.tar.gz (149.7MB)
    100% |████████████████████████████████| 149.7MB 1.9kB/s
  Installing build dependencies ... error
  Complete output from command /usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-oehw0kcn --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel scikit-build cmake pip "numpy==1.13.3; python_version=='3.6' and platform_machine != 'aarch64'" "numpy==1.19.3; python_version>='3.6' and sys_platform == 'linux' and platform_machine == 'aarch64'" "numpy==1.20.1; python_version>='3.6' and sys_platform == 'darwin' and platform_machine == 'aarch64'" "numpy==1.14.5; python_version=='3.7' and platform_machine != 'aarch64'" "numpy==1.17.3; python_version=='3.8' and platform_machine != 'aarch64'" "numpy==1.19.3; python_version>='3.9' and platform_machine != 'aarch64'":
  Ignoring numpy: markers 'python_version == "3.6" and platform_machine != "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version >= "3.6" and sys_platform == "darwin" and platform_machine == "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.7" and platform_machine != "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.8" and platform_machine != "aarch64"' don't match your environment
  Ignoring numpy: markers 'python_version >= "3.9" and platform_machine != "aarch64"' don't match your environment
  Collecting setuptools
    Retrying (Retry(total=4, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f91cda828>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/setuptools/
    Retrying (Retry(total=3, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f91cbd438>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/setuptools/
    Retrying (Retry(total=2, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f91cbd940>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/setuptools/
    Retrying (Retry(total=1, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f91ed0978>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/setuptools/
    Retrying (Retry(total=0, connect=None, read=None, redirect=None, status=None)) after connection broken by 'NewConnectionError('<urllib3.connection.VerifiedHTTPSConnection object at 0x7f91ed04e0>: Failed to establish a new connection: [Errno -3] Temporary failure in name resolution')': /simple/setuptools/
    Could not find a version that satisfies the requirement setuptools (from versions: )
  No matching distribution found for setuptools

  ----------------------------------------
Command "/usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-oehw0kcn --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel scikit-build cmake pip "numpy==1.13.3; python_version=='3.6' and platform_machine != 'aarch64'" "numpy==1.19.3; python_version>='3.6' and sys_platform == 'linux' and platform_machine == 'aarch64'" "numpy==1.20.1; python_version>='3.6' and sys_platform == 'darwin' and platform_machine == 'aarch64'" "numpy==1.14.5; python_version=='3.7' and platform_machine != 'aarch64'" "numpy==1.17.3; python_version=='3.8' and platform_machine != 'aarch64'" "numpy==1.19.3; python_version>='3.9' and platform_machine != 'aarch64'"" failed with error code 1 in None
```

manoj7410, 2021-06-02:
Can you try with the different versions of python3-opencv mentioned in comment https://github.com/google-coral/examples-camera/issues/76#issuecomment-852625906 ?

alengcm, 2021-06-03:
In the detect.py. Camera index should be 1. The default in the script is 0.  
After that there's another issue which I encountered on cap = cv2.VideoCapture('/dev/video1'): VIDIOC_QUERYCTRL: Inappropriate ioctl for device.

SunBeenMoon, 2021-06-03:
I did the same thing with @alengcm 
I've the these error code. 
```
mendel@silly-quill:~/google-coral/examples-camera/opencv$ python3 detect.py
Loading ../all_models/mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite with ../all_models/coco_labels.txt labels.
VIDIOC_QUERYCTRL: Inappropriate ioctl for device
VIDIOC_QUERYCTRL: Inappropriate ioctl for device
VIDIOC_QUERYCTRL: Inappropriate ioctl for device
```
[image]

SunBeenMoon, 2021-06-03:
I've tried all of the version wrote in  #76 (comment) 
But, It does not work. 
A viewed as an example here.

```
mendel@silly-quill:~/google-coral/examples-camera/opencv$ sudo pip3 install opencv-contrib-python==3.4.11.45
Collecting opencv-contrib-python==3.4.11.45
  Downloading https://files.pythonhosted.org/packages/34/99/0bd894f21fb6ee7618a6fca5002ea5530d54028838e8ee63430bd47eb745/opencv-contrib-python-3.4.11.45.tar.gz (142.5MB)
    100% |████████████████████████████████| 142.5MB 2.0kB/s
  Installing build dependencies ... error
  Complete output from command /usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-cwk2u8nw --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel scikit-build cmake pip "numpy==1.13.3; python_version=='3.6'" "numpy==1.14.5; python_version=='3.7'" "numpy==1.17.3; python_version=='3.8'" "numpy==1.19.3; python_version>='3.9'":
  Ignoring numpy: markers 'python_version == "3.6"' don't match your environment
  Ignoring numpy: markers 'python_version == "3.8"' don't match your environment
  Ignoring numpy: markers 'python_version >= "3.9"' don't match your environment
  Collecting setuptools
    Using cached https://files.pythonhosted.org/packages/4e/78/56aa1b5f4d8ac548755ae767d84f0be54fdd9d404197a3d9e4659d272348/setuptools-57.0.0-py3-none-any.whl
  Collecting wheel
    Using cached https://files.pythonhosted.org/packages/65/63/39d04c74222770ed1589c0eaba06c05891801219272420b40311cd60c880/wheel-0.36.2-py2.py3-none-any.whl
  Collecting scikit-build
    Using cached https://files.pythonhosted.org/packages/78/c9/7c2c7397ea64e36ebb292446896edcdecbb8c1aa6b9a1a32f6f67984c3df/scikit_build-0.11.1-py2.py3-none-any.whl
  Collecting cmake
    Using cached https://files.pythonhosted.org/packages/0a/37/2ff538a860a694780e1f2150e0616f7202c15a5dde0833ee09050002c091/cmake-3.20.2.tar.gz
    Installing build dependencies: started
    Installing build dependencies: finished with status 'done'
  Collecting pip
    Using cached https://files.pythonhosted.org/packages/cd/82/04e9aaf603fdbaecb4323b9e723f13c92c245f6ab2902195c53987848c78/pip-21.1.2-py3-none-any.whl
  Collecting numpy==1.14.5
    Downloading https://files.pythonhosted.org/packages/d5/6e/f00492653d0fdf6497a181a1c1d46bbea5a2383e7faf4c8ca6d6f3d2581d/numpy-1.14.5.zip (4.9MB)
  Collecting packaging (from scikit-build)
    Using cached https://files.pythonhosted.org/packages/3e/89/7ea760b4daa42653ece2380531c90f64788d979110a2ab51049d92f408af/packaging-20.9-py2.py3-none-any.whl
  Collecting distro (from scikit-build)
    Using cached https://files.pythonhosted.org/packages/25/b7/b3c4270a11414cb22c6352ebc7a83aaa3712043be29daa05018fd5a5c956/distro-1.5.0-py2.py3-none-any.whl
  Collecting pyparsing>=2.0.2 (from packaging->scikit-build)
    Using cached https://files.pythonhosted.org/packages/8a/bb/488841f56197b13700afd5658fc279a2025a39e22449b7cf29864669b15d/pyparsing-2.4.7-py2.py3-none-any.whl
  Building wheels for collected packages: cmake, numpy
    Running setup.py bdist_wheel for cmake: started
    Running setup.py bdist_wheel for cmake: finished with status 'error'
    Complete output from command /usr/bin/python3 -u -c "import setuptools, tokenize;__file__='/tmp/pip-install-k0ga0ppn/cmake/setup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdist_wheel -d /tmp/pip-wheel-9zmbzhmx --python-tag cp37:
    CMake Error: Could not find CMAKE_ROOT !!!
    CMake has most likely not been installed correctly.
    Modules directory not found in
    /usr/local/share/cmake-3.17

    --------------------------------------------------------------------------------
    -- Trying "Ninja" generator
    --------------------------------
    ---------------------------
    ----------------------
    -----------------
    ------------
    -------
    --
    Not searching for unused variables given on the command line.
    CMake Error: Could not find CMAKE_ROOT !!!
    CMake has most likely not been installed correctly.
    Modules directory not found in
    /usr/local/share/cmake-3.17
    CMake Error: Error executing cmake::LoadCache(). Aborting.

    --
    -------
    ------------
    -----------------
    ----------------------
    ---------------------------
    --------------------------------
    -- Trying "Ninja" generator - failure
    --------------------------------------------------------------------------------

    --------------------------------------------------------------------------------
[... 464 lines cut ...]
      --------------------------------------------------------------------------------

      ********************************************************************************
      scikit-build could not get a working generator for your system. Aborting build.

      Building Linux wheels for Python 3.7 requires a compiler (e.g gcc).
      But scikit-build does *NOT* know how to install it on mendel

      To build compliant wheels, consider using the manylinux system described in PEP-513.
      Get it with "dockcross/manylinux-x64" docker image:

        https://github.com/dockcross/dockcross#readme

      For more details, please refer to scikit-build documentation:

        http://scikit-build.readthedocs.io/en/latest/generators.html#linux

      ********************************************************************************

[... cut ...] all-k0ga0ppn/cmake/

  ----------------------------------------
Command "/usr/bin/python3 -m pip install --ignore-installed --no-user --prefix /tmp/pip-build-env-cwk2u8nw --no-warn-script-location --no-binary :none: --only-binary :none: -i https://pypi.org/simple -- setuptools wheel scikit-build cmake pip "numpy==1.13.3; python_version=='3.6'" "numpy==1.14.5; python_version=='3.7'" "numpy==1.17.3; python_version=='3.8'" "numpy==1.19.3; python_version>='3.9'"" failed with error code 1 in No
```ne

alengcm, 2021-06-03:
@SunBeenMoon run these

sudo apt-get update

sudo apt-get dist-upgrade

sudo reboot now 

Then install cmake and try again.

SunBeenMoon, 2021-06-03:
Hi @alengcm 
I did all of the command that you suggest me. 
But the error code is same as 

> I've tried all of the version wrote in #76 (comment)
> But, It does not work.
> A viewed as an example here.
>
> ```
> mendel@silly-quill:~/google-coral/examples-camera/opencv$ sudo pip3 install [...]

alengcm, 2021-06-03:
@SunBeenMoon it says you don't have cmake. Maybe remove it and install..  
sudo apt-get remove cmake
sudo apt-get install cmake

chadwallacehart, 2021-06-03:
I ran into the same issues. I struggled trying to install various versions via pip and finally got it to work by building openCV from scratch. I used the script here for that: https://github.com/milq/milq/blob/master/scripts/bash/install-opencv.sh

Note: There was a file instead of a directory at `/usr/local/share` - I'm not sure where that came from but I had to remove it and make sure there was a `/usr/local/share/licenses` directory for the `make install` part of the script.

WARNING - the build process took me 12+ hours (maybe more since I stopped monitoring it and went to bed). There were a few points where it looked like thought it froze since no progress was reported for more than an hour, but it kept going.  Run this locally or use something like screen - my ssh connection broke and I had to start over. 

This installed OpenCV 4.5.1.
I also had to set the camera index to `1` as @alengcm mentioned.

Here is what I ran.

```
cd ~
wget https://raw.githubusercontent.com/milq/milq/master/scripts/bash/install-opencv.sh
chmod u+x install-opencv.sh
ls /usr/local/share/licenses
# make sure this directory exists - /usr/local/share was a text file for me, so I renamed that
# mv /usr/local/share /usr/local/share.bak
# mkdir /usr/local/share
# mkdir /usr/local/share/licenses
./install.sh
cd coral/examples-camera/opencv
# make sure to set the display if you ssh in
DISPLAY=:0 python3 detect.py --camera_idx 1
```

Request for the Coral team: please fix the build in the next Mendel release to avoid this process.

alengcm, 2021-06-04:
@chadwallacehart @SunBeenMoon 
I was able to install the headless version using 
`pip3 install opencv-python-headless` 

I allocated extra 1gb of swap memory and had to install openssl and cmake. Took 7 hours.

SunBeenMoon, 2021-06-08:
Thank you @chadwallacehart 
I did as you explained and the problem was solved!!

closed 2021-06-08 by SunBeenMoon

monocle-dominic, 2021-06-17:
Thanks for sharing your solution @chadwallacehart 
Fixed the problem for me as well!

Marenz, 2023-04-18:
For me it wasn't as easy and it took me many many many hours to get it finally working. And to make the pain less for future people, here is the right numpy with the right opencv-python-headless wheels for you to install.
Just repeating it for the lazy reader what this is:

TO FIX 
```
VIDIOC_QUERYCTRL: Inappropriate ioctl for device
VIDIOC_QUERYCTRL: Inappropriate ioctl for device
VIDIOC_QUERYCTRL: Inappropriate ioctl for device
```

USE THESE WHEELS:

[opencv-python-headless.zip](https://github.com/google-coral/examples-camera/files/11266933/opencv-python-headless.zip)
[numpy-1.19.5.zip](https://github.com/google-coral/examples-camera/files/11266939/numpy-1.19.5.zip)

Marenz, 2023-04-19:
And because I had the build environment already setup, here is the non-headless version.

[opencv-python.zip](https://github.com/google-coral/examples-camera/files/11270998/opencv-python.zip)