google-coral/examples-camera #79: IndexError: list index out of range

fablau, 2021-06-10:
I am trying the mobilenet model with the command below:

`python3 detect.py --camera_idx 3 --model ../all_models/mobilenet_v2_1.0_224_quant_edgetpu.tflite --labels ../all_models/imagenet_labels.txt`

But I get the following error:

`Traceback (most recent call last):
  File "detect.py", line 99, in <module>
    main()
  File "detect.py", line 72, in main
    objs = get_objects(interpreter, args.threshold)[:args.top_k]
  File "/Users/fabrizio/.local/share/virtualenvs/Coral-u4jlFv0K/lib/python3.8/site-packages/pycoral/adapters/detect.py", line 190, in get_objects
    class_ids = common.output_tensor(interpreter, 1)[0]
  File "/Users/fabrizio/.local/share/virtualenvs/Coral-u4jlFv0K/lib/python3.8/site-packages/pycoral/adapters/common.py", line 29, in output_tensor
    return interpreter.tensor(interpreter.get_output_details()[i]['index'])()
IndexError: list index out of range`

Same issues with all other models with the only exception of the coco model. Any ideas?

manoj7410, 2021-06-16:
@fablau Can you share the environment details to reproduce this ?

fablau, 2021-06-16:
Thank you for your reply. At the end, I figured out that I was probably trying to perform detection with classification models. Do you think that may be the case?

Anyway, I am working inside a virtual environment via pipenv on MacOS 10.15. Thanks.

manoj7410, 2021-06-16:
@fablau That can be a reason. Is your demo working fine now ?

fablau, 2021-06-16:
Yes if I follow what the instructions say: for detection, I can just use the SSD models. Thank you!

closed 2021-06-17 by manoj7410

staebchen0, 2021-09-21:
I get the same error

list index out of range
Codezeile:
`objs = get_objects (interpreter, args.threshold) [: args.top_k]`

i created my model with a teachable machine - unquant Model
then i complied the model with the Edge TPU compiler

`edgetpu_compiler -s -m 13 model_unquant_8020.tflite`

do I have to export the teachable machine model as a quantized model so that it works?

operating system: Windows 10
Python 3.7
edgetpu_runtime_20210119  (the latest version does not work under Win10. Therefore I have to compile the model in runtime version 13 as well [https://github.com/google-coral/libedgetpu/issues/29#issuecomment-909207700](url)