google-coral/examples-camera #82: CoralDevBoard + Coral camera + streaming the detect results

juvinski, 2021-07-13:
Hi guys,

I have a coraldev board and coral camera, and how can I stream or save the video from gstreamer/detect.py samples to a file or stream to my desktop computer ?

Thanks a lot - I saw the same question but the answers I believe was using opencv or raspberry + coral accelerator.

zain-altaf, 2022-11-02:
Hi @juvinski were you able to figure out how to do it? I'm currently on the dev board mini and have a coral camera as well (mini and regular are basically the same software-wise). I'm trying to get it to save videos as well. Please let me know if you got it. 

Thanks!

nadavo11, 2022-12-13:
hey, just to check up on that topic, is there a way to do it?