google-coral/examples-camera #83: detect.py run as sudo

kiteklan, 2021-08-31:
### Description

I am trying to use detect.py code with some functions of environmental board.
Environmental Board requires GPIO permissions as root ,so I need the run with sudo.

`sudo python3  /home/mendel/google-coral/examples-camera/gstreamer/detect_1.py --model /home/mendel/one_nn11_edgetpu.tflite --labels /home/mendel/one_nn_label.txt --videosrc /dev/video1`

I am totally newbie , and Mendel has not got so many sources like raspberry.

Can someone lead me to right solution?

Either I need to give 'Mendel' user GPIO permissions or manage to run the script as sudo .

### Issue Type

Bug

### Coral Device

Dev Board, Enviro Board

### Programming Language

Python 3.6

### Relevant Log Output

```shell
Gstreamer pipeline:
 v4l2src device=/dev/video1 ! video/x-raw,width=640,height=480,framerate=30/1 ! decodebin ! glupload ! tee name=t
                t. ! queue ! glfilterbin filter=glbox name=glbox ! video/x-raw,format=RGB,width=320,height=320 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
                t. ! queue ! glsvgoverlaysink name=overlaysink
            

(detect_1.py:4694): Gtk-CRITICAL **: 10:55:23.943: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(detect_1.py:4694): Gtk-CRITICAL **: 10:55:23.943: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(detect_1.py:4694): Gtk-CRITICAL **: 10:55:23.943: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
Segmentation fault
```

hjonnala, 2021-08-31:
Hello @kiteklan  were you able to run this demo: https://coral.ai/docs/enviro-board/get-started/#3-try-the-demo

kiteklan, 2021-08-31:
Hi , yes i did .but I am using Coral Dev board requires root access to
control GPIO , I could not figure out to run as non root the demo.
Since the detect.py is also giving an error when running it with sudo.

kiteklan, 2021-08-31:
[detect_1.txt](https://github.com/google-coral/examples-camera/files/7084715/detect_1.txt)

when I run the code detect.py and environment sensor demo separately it works.
but I need the combine both functions under one file so I combined them .

**When running with sudo detect.py gives the error below:**

sudo python3  /home/mendel/google-coral/examples-camera/gstreamer/detect_1.py --model /home/mendel/one_nn11_edgetpu.tflite --labels /home/mendel/one_nn_label.txt --videosrc /dev/video1
Unable to init server: Could not connect: Connection refused
Unable to init server: Could not connect: Connection refused
Loading /home/mendel/one_nn11_edgetpu.tflite with /home/mendel/one_nn_label.txt labels.
Gstreamer pipeline:
 v4l2src device=/dev/video1 ! video/x-raw,width=640,height=480,framerate=30/1 ! decodebin ! glupload ! tee name=t
                t. ! queue ! glfilterbin filter=glbox name=glbox ! video/x-raw,format=RGB,width=320,height=320 ! appsink name=appsink emit-signals=true max-buffers=1 drop=true
                t. ! queue ! glsvgoverlaysink name=overlaysink
            

(detect_1.py:4064): Gtk-CRITICAL **: 14:22:52.310: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(detect_1.py:4064): Gtk-CRITICAL **: 14:22:52.311: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed

(detect_1.py:4064): Gtk-CRITICAL **: 14:22:52.311: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
Segmentation fault

**_When running without sudo this time the environmental sensor is failing,_**

python3  /home/mendel/google-coral/examples-camera/gstreamer/detect_1.py --model /home/mendel/one_nn11_edgetpu.tflite --labels /home/mendel/one_nn_label.txt --videosrc /dev/video1
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/periphery/gpio.py", line 88, in _open
    with open("/sys/class/gpio/gpio%d/direction" % pin, "w") as f_direction:
PermissionError: [Errno 13] Permission denied: '/sys/class/gpio/gpio138/direction'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "/home/mendel/google-coral/examples-camera/gstreamer/detect_1.py", line 145, in <module>
    main()
  File "/home/mendel/google-coral/examples-camera/gstreamer/detect_1.py", line 106, in main
    enviro = EnviroBoard()
  File "/usr/lib/python3/dist-packages/coral/enviro/board.py", line 69, in __init__
    self._display = ssd1306(serial_interface=spi(gpio=pGPIO(), port=port, device=0, gpio_DC=138, gpio_RST=140),
  File "/usr/lib/python3/dist-packages/luma/core/interface/serial.py", line 261, in __init__
    bitbang.__init__(self, gpio, transfer_size, DC=gpio_DC, RST=gpio_RST)
  File "/usr/lib/python3/dist-packages/luma/core/interface/serial.py", line 159, in __init__
    self._DC = self._configure(kwargs.get("DC"))
  File "/usr/lib/python3/dist-packages/luma/core/interface/serial.py", line 170, in _configure
    self._gpio.setup(pin, self._gpio.OUT)
  File "/usr/lib/python3/dist-packages/coral/enviro/rpi_gpio_periphery.py", line 33, in setup
    self._gpios[channels] = GPIO(channels, mode)
  File "/usr/lib/python3/dist-packages/periphery/gpio.py", line 43, in __init__
    self._open(pin, direction)
  File "/usr/lib/python3/dist-packages/periphery/gpio.py", line 91, in _open
    raise GPIOError(e.errno, "Setting GPIO direction: " + e.strerror)
periphery.gpio.GPIOError: [Errno 13] Setting GPIO direction: Permission denied

I need to overlay temp and humidity on the detection screen.

manoj7410, 2021-09-01:
@kiteklan Is this issue different from https://github.com/google-coral/edgetpu/issues/457 ?

kiteklan, 2021-09-01:
> @kiteklan Is this issue different from https://github.com/google-coral/edgetpu/issues/457 ?

@manoj7410 same problem but different side, I am trying to understand how to run the detect.py with elevated privileges, 'sudo phthon3 detect.py ...' results
With  
(detect_1.py:4064): Gtk-CRITICAL **: 14:22:52.311: _gtk_style_provider_private_get_settings: assertion 'GTK_IS_STYLE_PROVIDER_PRIVATE (provider)' failed
Segmentation fault

Being able run the script as root will help me to eliminate all GPIO and other permission problems.

hjonnala, 2021-09-12:
@kiteklan I am able to run detect_1.py as sudo. Can you please try the command below.

`sudo -E python3 detect_1.py --videosrc /dev/video1`

manoj7410, 2021-09-16:
@kiteklan Can this also be closed now ?

kiteklan, 2021-09-16:
Thanks the support, giving permission to GPIO pins solved the problem.

şunu yazdı:

closed 2021-09-16 by manoj7410