google-coral/examples-camera #84: FFMPEG to v4l2loopback  Failed to allocate a buffer

kiteklan, 2021-09-05:
### Description

I am trying to direct an h264 DVR stream to /dev/video2 with the command below.

`gst-launch-1.0 rtspsrc location="rtsp://admin:xxxxxxx@192.168.1.105:554/cam/realmonitor?channel=2&subtype=0" ! rtph264depay ! h264parse ! decodebin ! videoconvert ! video/x-raw,format=YUY2 ! tee ! v4l2sink device=/dev/video2`

it runs and the RTSP sources directed to the dummy device /dev/video0 below is the outcome

> Setting pipeline to PAUSED ...
> Pipeline is live and does not need PREROLL ...
> Progress: (open) Opening Stream
> Progress: (connect) Connecting to rtsp://admin:xxxxx@192.168.1.105:554/cam/realmonitor?channel=2&subtype=0
> Progress: (open) Retrieving server options
> Progress: (open) Retrieving media info
> Progress: (request) SETUP stream 0
> Progress: (open) Opened Stream
> Setting pipeline to PLAYING ...
> New clock: GstSystemClock
> Progress: (request) Sending PLAY request
> Progress: (request) Sending PLAY request
> Progress: (request) Sent PLAY request
> 

I ran the detection command below to get detection results from the NVR 
`edgetpu_detect --source /dev/video2:YUY2:960x1080:30/1 --model /home/mendel/ssdlite_mobiledet_mask_edgetpu.tflite  --labels /home/mendel/mask_labels.txt --threshold .30 --max_area .1 --min_area .01`

it works for a second ,display show the camera stream and  error code below occurs. 

> v4l2src device=/dev/video2 ! video/x-raw,format=YUY2,width=960,height=1080,framerate=30/1 ! glupload ! tee name=t
> t. ! queue ! glsvgoverlaysink name=glsink
> t. ! queue max-size-buffers=1 leaky=downstream ! glfilterbin filter=glbox ! video/x-raw,format=RGB,width=320,height=320 ! appsink name=appsink emit-signals=True max-buffers=1 drop=True sync=False
> Error: gst-resource-error-quark: Failed to allocate a buffer (14): gstv4l2src.c(998): gst_v4l2src_create (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0

I tried playing with the buffers of the sudo modprobe v4l2loopback max_buffers=x but could no. success.

Can somebody tell me what am I doing wrong.?

### Issue Type

Bug

### Operating System

Mendel Linux

### Coral Device

Dev Board

### Programming Language

Python 3.7

### Relevant Log Output

```shell
mendel@bored-snail:~$ gst-launch-1.0 rtspsrc location="rtsp://admin:xxxxx@192.168.1.105:554/cam/realmonitor?channel=2&subtype=0" ! rtph264depay ! h264parse ! decodebin ! videoconvert ! video/x-raw,format=YUY2 ! tee ! v4l2sink device=/dev/video2
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
Progress: (open) Opening Stream
Progress: (connect) Connecting to rtsp://admin:xxxxx@192.168.1.105:554/cam/realmonitor?channel=2&subtype=0
Progress: (open) Retrieving server options
Progress: (open) Retrieving media info
Progress: (request) SETUP stream 0
Progress: (open) Opened Stream
Setting pipeline to PLAYING ...
New clock: GstSystemClock
Progress: (request) Sending PLAY request
Progress: (request) Sending PLAY request
Progress: (request) Sent PLAY request

mendel@bored-snail:~$ edgetpu_detect --source /dev/video2:YUY2:960x1080:30/1 --model /home/mendel/ssdlite_mobiledet_mask_edgetpu.tflite  --labels /home/mendel/mask_labels.txt --threshold .30 --max_area .1 --min_area .01
v4l2src device=/dev/video2 ! video/x-raw,format=YUY2,width=960,height=1080,framerate=30/1 ! glupload ! tee name=t
t. ! queue ! glsvgoverlaysink name=glsink
t. ! queue max-size-buffers=1 leaky=downstream ! glfilterbin filter=glbox ! video/x-raw,format=RGB,width=320,height=320 ! appsink name=appsink emit-signals=True max-buffers=1 drop=True sync=False
Error: gst-resource-error-quark: Failed to allocate a buffer (14): gstv4l2src.c(998): gst_v4l2src_create (): /GstPipeline:pipeline0/GstV4l2Src:v4l2src0
```

hjonnala, 2021-12-02:
@kiteklan can you try this [script](https://github.com/google-coral/examples-camera/blob/master/gstreamer/detect.py) with --videofmt [h264 flag](https://github.com/google-coral/examples-camera/blob/master/gstreamer/gstreamer.py#L211).

Watermelonseeds, 2022-07-08:
Hi,man!
I have the same problem, how did you solve it?