google-coral/examples-camera #85: gstreamer detect.py and classify.py user_callback function

suyash-narain, 2021-11-30:
Hi,

I needed some help in understanding how user_callback function. 
In classify.py, the function user_callback takes in "input_tensor, src_size, inference_box" arguments. If i replace "run_interpeter" by "interpreter.invoke", i can get rid of "input tensor" arg as well. 

But when this function is called in gstreamer.run_pipeline, no arguments are given to the function. I believe there are no defaults given to this function during its definition as well. Then how is this function working?

same way in detect.py as well?

thanks

hjonnala, 2021-11-30:
Hi @suyash-narain argumnets will be passed when invoking the user_callback function. Here is the flow:
https://github.com/google-coral/examples-camera/blob/3f3556eca31d9de656dbe26da72ceb8297870326/gstreamer/gstreamer.py#L205
https://github.com/google-coral/examples-camera/blob/3f3556eca31d9de656dbe26da72ceb8297870326/gstreamer/gstreamer.py#L277
https://github.com/google-coral/examples-camera/blob/3f3556eca31d9de656dbe26da72ceb8297870326/gstreamer/gstreamer.py#L278
https://github.com/google-coral/examples-camera/blob/3f3556eca31d9de656dbe26da72ceb8297870326/gstreamer/gstreamer.py#L127

suyash-narain, 2021-12-01:
thank you for your reply. I was wondering, If i use the same code on a rpi 4, but without the coral accelerator since I am running a simple mobilenet_v2_quant.tflite from tfhub, the detections and classifications do not happen. I get just one frame and the detections do not run in a loop, if i follow the codebase as you have provided. Do you have any suggestions how to change the gstreamer pipeline so as to run detect.py with a rpi4 without the coral accelerator if i want to run a tflite model not optimized for edge tpu?

since all the pycoral.adapter functions can be used with rpi as well since they are basic numpy based functions sourcing tflite.interpreter in most cases, what other change would i require to do to make it working with a normal rpi?

thanks in advance

hjonnala, 2021-12-02:
please try with the below changes to run CPU tflite model:

add the below import statement: 
`import tflite_runtime.interpreter as tflite`

and change this line as below: https://github.com/google-coral/examples-camera/blob/master/gstreamer/classify.py#L65
`tflite.Interpreter(model_path=args.model)
`

suyash-narain, 2021-12-03:
Hi,
I tried following what you suggested, and using the master branch, the classification results don't seem to get updated on the loop. no matter the usb camera is open or closed, i get the following output at all times:

  Inference: 36.40 ms FPS: 23 fps score=5.34: matchstick score=4.45: spotlight, spot score=4.05: nematode, nematode worm, roundworm
  Inference: 28.61 ms FPS: 23 fps score=5.34: matchstick score=4.45: spotlight, spot score=4.05: nematode, nematode worm, roundworm
  Inference: 57.90 ms FPS: 23 fps score=5.34: matchstick score=4.45: spotlight, spot score=4.05: nematode, nematode worm, roundworm
  Inference: 36.10 ms FPS: 23 fps score=5.34: matchstick score=4.45: spotlight, spot score=4.05: nematode, nematode worm, roundworm
  Inference: 52.16 ms FPS: 23 fps score=5.34: matchstick score=4.45: spotlight, spot score=4.05: nematode, nematode worm, roundworm
  Inference: 36.62 ms FPS: 23 fps score=5.34: matchstick score=4.45: spotlight, spot score=4.05: nematode, nematode worm, roundworm
Inference: 46.81 ms FPS: 23 fps score=5.34: matchstick score=4.45: spotlight, spot score=4.05: nematode, nematode worm, roundworm

the output seem not to get updated on for a video stream, and again, these outputs are totally wrong as well since there is no worm in the video stream to classify. 

same case with detection as well. the output boxes seem to get stuck and seem like a drawing superimposed on the image, not changing the detections with the changing frame. 

the only change in the code i did was to create a python script having all the functions used by pycorals since i was having issues installing pycorals on my rpi. 

the script used was:
#########################################################

"""Utilities to help process a dataset."""

import re
import tflite_runtime.interpreter as tflite

def read_label_file(file_path):
  """Reads labels from a text file and returns it as a dictionary.

  This function supports label files with the following formats:

  + Each line contains id and description separated by colon or space.
    Example: ``0:cat`` or ``0 cat``.
  + Each line contains a description only. The returned label id's are based on
    the row number.

  Args:
    file_path (str): path to the label file.

  Returns:
    Dict of (int, string) which maps label id to description.
  """
  with open(file_path, 'r', encoding='utf-8') as f:
    lines = f.readlines()
  ret = {}
  for row_number, content in enumerate(lines):
    pair = re.split(r'[:\s]+', content.strip(), maxsplit=1)
    if len(pair) == 2 and pair[0].strip().isdigit():
      ret[int(pair[0])] = pair[1].strip()
    else:
      ret[row_number] = content.strip()
  return ret

def make_interpreter(model_path_or_content, device=None, delegate=None):
  """Creates a new ``tf.lite.Interpreter`` instance using the given model.
  **Note:** If you have multiple Edge TPUs, you should always specify the
  ``device`` argument.
  Args:
     model_path_or_content (str or bytes): `str` object is interpreted as
       model path, `bytes` object is interpreted as model content.
     device (str): The Edge TPU device you want:
       + None      -- use any Edge TPU (this is the default)
       + ":<N>"    -- use N-th Edge TPU (this corresponds to the enumerated
         index position from :func:`list_edge_tpus`)
       + "usb"     -- use any USB Edge TPU
       + "usb:<N>" -- use N-th USB Edge TPU
       + "pci"     -- use any PCIe Edge TPU
       + "pci:<N>" -- use N-th PCIe Edge TPU
       If left as None, you cannot reliably predict which device you'll get.
       So if you have multiple Edge TPUs and want to run a specific model on
       each one, then you must specify the device.
[... 260 lines cut ...]
def get_objects(interpreter,
                score_threshold=-float('inf'),
                image_scale=(1.0, 1.0)):

  scores = output_tensor(interpreter, 0)[0]
  boxes = output_tensor(interpreter, 1)[0]
  count = (int)(output_tensor(interpreter, 2)[0])
  class_ids = output_tensor(interpreter, 3)[0]

  width, height = input_size(interpreter)
  image_scale_x, image_scale_y = image_scale
  sx, sy = width / image_scale_x, height / image_scale_y

  def make(i):
    ymin, xmin, ymax, xmax = boxes[i]
    return Object(
        id=int(class_ids[i]),
        score=float(scores[i]),
        bbox=BBox(xmin=xmin, ymin=ymin, xmax=xmax,
                  ymax=ymax).scale(sx, sy).map(int))

  return [make(i) for i in range(count) if scores[i] >= score_threshold]
#############################################################

not entirely sure why am i not getting correct classifications and detections and they are also not changing with each passing frame in the live video stream. I don't have any issues while running it on opencv though

hjonnala, 2021-12-03:
Please try to install the pycoral 2.0 and tflite_runtime 2.5.0.Post1 using the command below and try the scripts:
 
`python3 -m pip install --extra-index-url https://google-coral.github.io/py-repo/ pycoral~=2.0`

If you are getting into any issues please share the error details.

suyash-narain, 2021-12-07:
I installed pycoral 2.0 and tflite_runtime 2.5.0.Post1 using the command mentioned and tried to run detect.py
but now I am getting this error:
"
AttributeError: module 'tflite_runtime.interpreter' has no attribute 'Interpreter'
"

I was not facing this issue with the earlier instance of tflite 2.4 installed on my board.

hjonnala, 2021-12-07:
can you please share this info:

```
hemanth@hemanth-glaptop:~$ python3
Python 3.9.7 (default, Oct 19 2021, 19:27:03) 
[GCC 10.3.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import tflite_runtime.interpreter as tflite
>>> dir(tflite)
['Delegate', 'Interpreter', 'InterpreterWithCustomOps', 'OpResolverType', 'SignatureRunner', '__builtins__', '__cached__', '__doc__', '__file__', '__loader__', '__name__', '__package__', '__spec__', '_get_op_resolver_id', '_interpreter_wrapper', '_tf_export', 'absolute_import', 'ctypes', 'division', 'enum', 'load_delegate', 'metrics', 'np', 'os', 'platform', 'print_function', 'sys']
>>> import tflite_runtime
>>> tflite_runtime.__version__
'2.5.0.post1'
>>> 
```
Please first try the classify demo with these changes: https://github.com/google-coral/examples-camera/issues/85#issuecomment-984700861

hjonnala, 2021-12-23:
Closing due to inactivity. Feel free to reopen if you have any questions. Thanks!

closed 2021-12-23 by hjonnala