google-coral/examples-camera #88: Undefined reference to `tflite::optimize::sparsity::FormatConverter

bsteiner-dandy, 2022-03-08:
### Description

Hi, following the instructions for building `libtensorflow-lite` for the `nativeapp` I came across this error. I also got this same error when attempting to build it on the coral instead. The TENSORFLOW_COMMIT I used was `d855adfc5a0195788bf5f92c3c7352e638aa1109`

### Issue Type

Build/Install

### Operating System

Mendel Linux, Linux

### Coral Device

Dev Board

### Programming Language

C++

### Relevant Log Output

```shell
usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: /home/brad/projects/tensorflow/tensorflow/lite/tools/make/gen/aarch64_armv8-a/lib/libtensorflow-lite.a(densify.o): in function `tflite::ops::builtin::densify::Eval(TfLiteContext*, TfLiteNode*)':
densify.cc:(.text+0x51c): undefined reference to `tflite::optimize::sparsity::FormatConverter<signed char>::FormatConverter(std::vector<int, std::allocator<int> > const&, TfLiteSparsity const&)'
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: densify.cc:(.text+0x528): undefined reference to `tflite::optimize::sparsity::FormatConverter<signed char>::SparseToDense(signed char const*)'
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: densify.cc:(.text+0x760): undefined reference to `tflite::optimize::sparsity::FormatConverter<float>::FormatConverter(std::vector<int, std::allocator<int> > const&, TfLiteSparsity const&)'
/usr/lib/gcc-cross/aarch64-linux-gnu/11/../../../../aarch64-linux-gnu/bin/ld: densify.cc:(.text+0x76c): undefined reference to `tflite::optimize::sparsity::FormatConverter<float>::SparseToDense(float const*)'
collect2: error: ld returned 1 exit status
```

bsteiner-dandy, 2022-03-09:
Getting a new error with that commit:

`In file included from ./tensorflow/lite/core/api/op_resolver.h:23,
                 from ./tensorflow/lite/c/c_api_internal.h:25,
                 from tensorflow/lite/c/c_api_experimental.cc:24:
./tensorflow/lite/schema/schema_generated.h: In function ‘const char* tflite::EnumNameTensorType(tflite::TensorType)’:
./tensorflow/lite/schema/schema_generated.h:470:20: error: ‘IsOutRange’ is not a member of ‘flatbuffers’
  470 |   if (flatbuffers::IsOutRange(e, TensorType_FLOAT32, TensorType_UINT32)) return "";
      |             `

manoj7410, 2022-03-14:
@bsteiner-dandy Can you share the steps to replicate this issue?

hjonnala, 2022-03-14:
@please try adding these the below lines to tensorflow/lite/tools/make/downloads/ruy/ruy/block_map.cc file:

#include <stdexcept>
#include <limits>

bsteiner-dandy, 2022-03-14:
Looks like it builds after that

hjonnala, 2022-03-21:
@bsteiner-dandy I think, we have modify some scripts in native app for TF2.5 commit. Are you able to run native app demo successfully?

bsteiner-dandy, 2022-03-21:
I have not tried yet. I have been using the opencv python approach for now

bsteiner-dandy, 2022-04-11:
I am seeing some issues with the python approach so wanted to try this approach again. However when I go to build the native app, I get this error:

```
inferencewrapper.cc: In constructor ‘coral::InferenceWrapper::InferenceWrapper(const string&, const string&)’:
inferencewrapper.cc:58:74: error: no matching function for call to ‘tflite::Interpreter::ModifyGraphWithDelegate(<brace-enclosed initializer list>)’
   interpreter_->ModifyGraphWithDelegate({delegate, edgetpu_free_delegate});
                                                                          ^
In file included from inferencewrapper.h:9,
                 from inferencewrapper.cc:1:
/home/mendel/sd_storage/tensorflow/tensorflow/lite/interpreter.h:489:16: note: candidate: ‘TfLiteStatus tflite::Interpreter::ModifyGraphWithDelegate(TfLiteDelegate*)’
   TfLiteStatus ModifyGraphWithDelegate(TfLiteDelegate* delegate);
                ^~~~~~~~~~~~~~~~~~~~~~~
/home/mendel/sd_storage/tensorflow/tensorflow/lite/interpreter.h:489:16: note:   no known conversion for argument 1 from ‘<brace-enclosed initializer list>’ to ‘TfLiteDelegate*’
/home/mendel/sd_storage/tensorflow/tensorflow/lite/interpreter.h:499:23: note: candidate: ‘template<class Delegate, class Deleter> TfLiteStatus tflite::Interpreter::ModifyGraphWithDelegate(std::unique_ptr<_Tp, _Dp>)’
   inline TfLiteStatus ModifyGraphWithDelegate(
                       ^~~~~~~~~~~~~~~~~~~~~~~
/home/mendel/sd_storage/tensorflow/tensorflow/lite/interpreter.h:499:23: note:   template argument deduction/substitution failed:
inferencewrapper.cc:58:74: note:   couldn't deduce template parameter ‘Delegate’
   interpreter_->ModifyGraphWithDelegate({delegate, edgetpu_free_delegate});
```

Namburger, 2022-04-11:
@bsteiner-dandy checkout this PR :)
https://github.com/google-coral/examples-camera/pull/89/files

closed 2022-04-12 by hjonnala