google-coral/examples-camera #92: nativeapp:  docker install fails with: unable to find "net_cls"

jlpoolen, 2022-05-10:
### Description

    mendel@coral1:~/coral/examples-camera$ cd nativeapp/
    mendel@coral1:~/coral/examples-camera/nativeapp$  make DOCKER_TARGETS=mendelcam DOCKER_CPUS=aarch64 docker-build
    docker build  -t "native-app-debian-buster" \
        --build-arg IMAGE=debian:buster /home/mendel/coral/examples-camera/nativeapp/docker
    Sending build context to Docker daemon  15.36kB
    Step 1/10 : ARG IMAGE
    Step 2/10 : FROM ${IMAGE}
    buster: Pulling from library/debian
    b9f330b3a7e2: Pull complete
    Digest: sha256:ebe4b9831fb22dfa778de4ffcb8ea0ad69b5d782d4e86cab14cc1fded5d8e761
    Status: Downloaded newer image for debian:buster
     ---> de70a5f02052
    Step 3/10 : COPY update_sources.sh /
     ---> bf4bbaa488f8
    Step 4/10 : RUN /update_sources.sh
     ---> Running in 3a46365427a6
    unable to find "net_cls" in controller set: unknown
    make: *** [/home/mendel/coral/examples-camera/nativeapp/docker/docker.mk:30: docker-image] Error 1
    mendel@coral1:~/coral/examples-camera/nativeapp$

### Issue Type

Build/Install

### Operating System

Mendel Linux

### Coral Device

Dev Board

### Relevant Log Output

```shell
See Description above.
```

Namburger, 2022-05-10:
So the docker build were really not supposed to be used on mendel (the dev board). The whole point of using docker is so that we can consistently cross compile this example on your robust machine with cpu target aarch64 and then copy the binary over to mendel. Using docker and all of the libraries needed will boat the board and for sure run out of ram to build.

jlpoolen, 2022-05-10:
I recommend you have something to the same effect as your explanation in your comment above in the README.md.  Perhaps a preface of:

This project is designed to be downloaded onto a workstation where the program will be cross-compiled within Docker for installation on the Google Coral Dev Board.  This is not intended for direct installation on the Coral Dev Board.