google-coral/examples-camera #97: How to actually run this thing

xrbeattx, 2022-09-15:
### Description

Hello I am really new to all of this and I am able to get the image classification example and the facial detection running with the Coral Camera, however I am trying to mimic the commands to get any other example running but I dont understand the command to do so.  These commands arent listed on the readme so I am just assuming its either edgetpu_detect or edgetpu_classify as their are not any other variable names that I can see on my system as of now.  
[image]
Also the README states to use this model mobilenet_ssd_v2_coco_quant_postprocess_edgetpu.tflite, coco_labels.txt but the only model I can git clone from their test_data page (https://github.com/google-coral/test_data) is ssd_mobilenet_v2 .... is this an error or am I (probably) just totally misunderstanding what this is? With my limited understanding of CV and Linux I am thinking I can mimic this command edgetpu_classify \
--model ${DEMO_FILES}/mobilenet_v2_1.0_224_quant_edgetpu.tflite \
--labels ${DEMO_FILES}/imagenet_labels.txt 
and replace the --model with ones I downloaded from the test_example page 
sudo wget -P ${DEMO_FILES}/ https://github.com/google-coral/test_data/blob/master/coco_labels.txt and
sudo wget -P ${DEMO_FILES}/ https://github.com/google-coral/test_data/blob/master/ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite

This seemed to not work so at this point I am just at a loss as to how to get this working with other examples Please help

### Issue Type

Support

### Operating System

Mendel Linux

### Programming Language

Python 3.7

### Relevant Log Output

```shell
Traceback (most recent call last):
  File "/usr/bin/edgetpu_detect", line 11, in <module>
    load_entry_point('edgetpuvision==7.0', 'console_scripts', 'edgetpu_detect')()
  File "/usr/lib/python3/dist-packages/edgetpuvision/detect.py", line 193, in main
    run_app(add_render_gen_args, render_gen)
  File "/usr/lib/python3/dist-packages/edgetpuvision/apps.py", line 70, in run_app
    display=args.displaymode):
  File "/usr/lib/python3/dist-packages/edgetpuvision/gstreamer.py", line 241, in run_gen
    inference_size = render_overlay_gen.send(None)  # Initialize.
  File "/usr/lib/python3/dist-packages/edgetpuvision/detect.py", line 128, in render_gen
    interpreters, titles = utils.make_interpreters(args.model)
  File "/usr/lib/python3/dist-packages/edgetpuvision/utils.py", line 52, in make_interpreters
    interpreter = edgetpu.make_interpreter(model_path)
  File "/usr/lib/python3/dist-packages/pycoral/utils/edgetpu.py", line 93, in make_interpreter
    model_path=model_path_or_content, experimental_delegates=delegates)
  File "/usr/lib/python3/dist-packages/tflite_runtime/interpreter.py", line 351, in __init__
    experimental_preserve_all_tensors))
ValueError: Model provided has model identifier '

<', should be 'TFL3'
```

hjonnala, 2022-09-15:
The link you are trying to download the model is not correct. Please try the below link:

wget https://github.com/google-coral/test_data/raw/master/ssd_mobilenet_v2_coco_quant_postprocess_edgetpu.tflite

xrbeattx, 2022-09-15:
Awesome thank you for such a quick response! In the meantime I got the Posenet working, that is really fun and easy to implement.

 By the speed and simplicity of your answer I assume that I am correct to use 'edgetpu_detect' --model etc etc like in the picture above? I will give it a try here again in a minute, just waiting for a mount to finish 3d printing

xrbeattx, 2022-09-19:
That worked! Thank you very much, overlay seems to blink out when over 20 ish objects are detected and boy howdy does it detect a lot of objects. No idea what these objects are but still a fun little toy to play with

closed 2022-09-19 by xrbeattx