google-coral/examples-camera #100: Raspberry Pi camera on Coral Edge Tpu board?

charbull, 2022-11-27:
### Description

Hi,

It seems that the Coral Camera for the Coral Dev Board is out of stock: https://coral.ai/products/camera/
Was curious if someone plugged a raspberry pi camera instead on a coral dev board? 
Such as this one? https://www.adafruit.com/product/3100

Cheers,
Charbull

### Issue Type

Others

### Operating System

Mendel Linux

### Coral Device

Dev Board

MarioCavero, 2022-12-20:
Will try it out soon, but my first guess is that it should, if the connector is MIPI-Csi. Coral:  Connects through the MIPI-CSI interface.

Raspberry pi camera: It attaches to the Pi by way of one of the small sockets on the board's upper surface and uses the dedicated CSi interface, designed especially for interfacing to cameras.

MxFxM, 2024-04-08:
Any results yet? I was stumbling across this question too.