google-coral/examples-camera #101: Can't donwload all_models.tar

blackCmd, 2022-12-01:
### Description

Error with `sh download_models.sh`

### Issue Type

Support

### Operating System

Ubuntu

### Coral Device

USB Accelerator

### Programming Language

Python 3.8

### Relevant Log Output

```shell
--2022-12-01 11:08:51--  https://dl.google.com/coral/canned_models/all_models.tar.gz
Resolving dl.google.com (dl.google.com)... 142.250.76.142, 2404:6800:400a:80b::200e
Connecting to dl.google.com (dl.google.com)|142.250.76.142|:443... connected.
ERROR: cannot verify dl.google.com's certificate, issued by ‘CN=GTS CA 1C3,O=Google Trust Services LLC,C=US’:
  Unable to locally verify the issuer's authority.
To connect to dl.google.com insecurely, use `--no-check-certificate'.
tar (child): all_models.tar.gz: Cannot open: No such file or directory
tar (child): Error is not recoverable: exiting now
tar: Child returned status 2
tar: Error is not recoverable: exiting now
```

hjonnala, 2022-12-01:
can you try with --no-check-certificate.. 
`wget https://dl.google.com/coral/canned_models/all_models.tar.gz --no-check-certificate`

dmitriykovalev, 2022-12-02:
Also please make sure your clock (date and time) is correct, the link works fine:

```
$ curl --head https://dl.google.com/coral/canned_models/all_models.tar.gz
HTTP/2 200 
accept-ranges: bytes
content-disposition: attachment
content-security-policy: default-src 'none'
server: downloads
x-content-type-options: nosniff
x-frame-options: SAMEORIGIN
x-xss-protection: 0
date: Fri, 02 Dec 2022 18:20:34 GMT
cache-control: public,max-age=86400
last-modified: Wed, 17 Apr 2019 17:41:12 GMT
etag: "378ec0"
content-type: application/gzip
content-length: 184961153
age: 0
alt-svc: h3=":443"; ma=2592000,h3-29=":443"; ma=2592000,h3-Q050=":443"; ma=2592000,h3-Q046=":443"; ma=2592000,h3-Q043=":443"; ma=2592000,quic=":443"; ma=2592000; v="46,43"
```

closed 2022-12-02 by dmitriykovalev